InsanityAutomation/Marlin #115: TM Info Screens after update issue

masc2279, 2020-04-03:
### Description

images and data on progress screen do not align and is missing items based on the background image. Looks like it still thinks the stock screens are there vs the CR-XABL_Screens_V2Rev1.7z screens as seen on image below

[image]

### Steps to Reproduce

1. get CrealityDwin_2.0
2. Configure for CR10XMax
3. upload firmware to printer
4. upload CR-XABL_Screens_V2Rev1.7z
5. Start Print

**Expected behavior:** Data should be present and aligned on screen

**Actual behavior:** Missing data and alignment is based on stock screens

InsanityAutomation, 2020-04-04:
It sounds like something went wrong packaging the CRX screens. Ill go over them this weekend.

InsanityAutomation, 2020-04-05:
Fixed, the uploaded archive had 2 bin files inside, the naming the dgus tool uses and what the screen looks for. Correct data was in the default export name and the screen used the old files! Regardless, fresh archive is loaded. If anything was missed, just reopen the issue.

closed 2020-04-05 by InsanityAutomation

masc2279, 2020-04-05:
Screens  look good opened another ticket up for CrealityDwin2.0_Bleeding   the sounds are all not working or broken uploaded audio recording as well on that ticket.  

Thanks for your help.

LMS0815, 2020-04-06:
I am facing the same issue. 
Where can I find the updated version?
Version `CR-XABL_Screens_V2Rev1.1 ` looks strange to me.

`/DWIN_SET/` contains 4 bin files and a sub folder (`14`) containing a file `bin` as well. I assume it should be `14.bin` and not `14/bin.`.

masc2279, 2020-04-06:
> I am facing the same issue.
> Where can I find `CR-XABL_Screens_V2Rev1.x.7z` version?
> Version `CR-XABL_Screens_V2Rev1.1.7z ` looks strange to me.
> 
> `/DWIN_SET/` contains 4 `*.bin` files and a sub folder `/14/` containing a file `bin.` as well. I assume it should be `14.bin` and not `14/bin.`.
> 
> Found: https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin2.0_Bleeding/CR-XABL_Screens_V2Rev2.7z

Just a heads up that the screens work in that file but there is a sound bug that i have opened on that you might want to set your sound down to mute .... But the screens do work at least on my machine now

LMS0815, 2020-04-06:
Thank you. 
@masc2279 I can not reproduce the sound bug with my CR-X (**CRX_BLT_UBL_Fil_DW6.hex**). I just needed to lower volume, which is 100% after update.