InsanityAutomation/Marlin #116: Sounds are broken CR-10MAX

masc2279, 2020-04-05:
[Printer Sounds.m4a.zip](https://github.com/InsanityAutomation/Marlin/files/4434575/Printer.Sounds.m4a.zip)

Steps to Reproduce

1. Get CrealityDwin2.0_Bleeding
2. Configure for CR10XMax
3. Upload firmware to printer
4. Upload CR-XABL_Screens_V2Rev2.7z to screen.

Boot printer

Expected behavior: Touch screen sounds should be normal and clear.

Actual behavior: Touch screen sounds and boot up sounds are broken and damaged. Large squeaks and scrapes sounds are heard.

rc-fire, 2020-04-08:
I just tried to install the tiny machines firmware on a fairly new CR10S Pro V2 and got the exact same issue after the LCD firmware update. Did you find a way to fix it or is there more information on this? (I am also unable to start auto bed leveling from the LCD while over gcode everything works fine - I just mention it here in case it might be related)

masc2279, 2020-04-08:
No sounds have not been fixed nor has someone started looking into it ....  As for leveling on the 10MAX it works like a charm.  I also went to non bleeding codebase using the new screens and it works minus the sounds so if your having issues just update the screens and use the normal 2.0 version branch for code

masc2279, 2020-04-08:
> I just tried to install the tiny machines firmware on a fairly new CR10S Pro V2 and got the exact same issue after the LCD firmware update. Did you find a way to fix it or is there more information [...]

No sounds have not been fixed nor has someone started looking into it .... As for leveling on the 10MAX it works like a charm. I also went to non bleeding codebase using the new screens and it works minus the sounds so if your having issues just update the screens and use the normal 2.0 version branch for code

barrarot, 2020-04-11:
Hi everyone, if it helps, I am also getting this issue of broken noises on my stock cr10s pro(May 2019) too, the following is what I did: 

1)installed http://insanityautomation.com/Firmware/Creality/SingleExtruderScreens_V2Rev1.1.7z dwin_set from tiny machines (https://tiny-machines-3d.myshopify.com/pages/10spro), which links to this repo, and installed it to the LCD. 

2)In addition installed http://insanityautomation.com/Firmware/Creality/10SPro_BIL_DW6.hex.gz from tiny machines, which also links to this repo. I just had to unzip the compression of .gz(to just a .hex), and installed through Cura and a usb connection.

pat931, 2020-04-11:
Hi everyone, if it helps, I am also getting this issue of broken noises on my stock cr10s pro(May 2019) too, the following is what I did:

1)installed http://insanityautomation.com/Firmware/Creality/SingleExtruderScreens_V2Rev1.1.7z dwin_set from tiny machines (https://tiny-machines-3d.myshopify.com/pages/10spro), which links to this repo, and installed it to the LCD.

2)In addition installed http://insanityautomation.com/Firmware/Creality/10SPro_BIL_DW6.hex.gz from tiny machines, which also links to this repo. I just had to unzip the compression of .gz(to just a .hex), and installed through Cura and a usb connection.

Myrddinwylit, 2020-04-13:
Like pat931 - I followed the Same directions as above. On my Stock Cr10s pro (v1 with Stock Probe)
The sound is not working properly. Turned off the Sound for Now.

pat931, 2020-04-13:
On that note from myrddinwylit above - It also resets my sounds settings every time I restart the machine. Anyway that when this issue gets resolved that also it could save the set sound level to persist beyond restart. If not maybe a feature request?

quarky42, 2020-04-24:
I've got no sound on my Ender 5 Plus.  I do have sound on my CR-10S Pro.

The first time I went to update the CR-10S Pro, I didn't know if I needed only the DWIN_SET folder or if I needed all the files from the Single Screen package so I just included all the files.

When I updated the Ender 5 Plus I just copied the DWIN_SET folder and nothing else.  

I'll see if I can get some time early next week to try copying all the files over on the Ender 5 Plus to see if that does the trick. 

Has anyone else tried this yet?

Myrddinwylit, 2020-05-04:
Any Updates on this Issue.. I am not sure if the fix is as simple as what quarky42 did. Where is the single screen package?

quarky42, 2020-05-04:
I haven't had a chance to take the bottom cover off and try some variations on updating the touch screen yet.  It is on my todo list, but I have had some other things keeping me busy.

MaltWhiskey, 2020-05-30:
i had the same issue, getting the wav files form an older screen version fixed the issue. Just replace all the wav files and update the screen again and done.

Myrddinwylit, 2020-06-01:
Can someone post the link to the older sound files - it’s not a major problem but would like to fix this

masc2279, 2020-06-09:
Just pulled todays latest   CrealityDwin2.0_Bleeding   branch which includes latest     bugfix-2.0.x   Bug and new fixes and this issue still exists and has not been resolved ....

clifjones, 2020-07-22:
I have a stock Cr10S Pro and had broken sound on DW6.  DW5 screen .WAV files didn't fix it but the B7 screen .WAV files did correct the issue.  Hopefully this helps others.

pat931, 2020-10-10:
Hi everyone, Im still having issues with sound on my cr10s pro stock version. Has anyone had any new luck with the newest release DW7? @clifjones you mention B7 screen .wav files and @MaltWhiskey you mention old .wav files. Is there a place I can download these? Thanks.

JarrettWendt, 2021-02-21:
I had this problem and it turned out that the default Archive Utility on MacOS was the culprit. If you're on a Mac, try unzipping with The Unarchiver or Keka instead.

pat931, 2021-02-21:
@JarrettWendt Thank you so much for the advise. I did a quick test on my Mac, and found the sounds work on my Mac after using The Unarchiver app with the 7z files. Before I was using the default Archive Utility. I'll push the new sound to my printer later, but I can verify the sounds are no longer broken and this fix works, as far as Mac goes.

InsanityAutomation, 2021-04-28:
> 
> 
> I had this problem and it turned out that the default Archive Utility on MacOS was the culprit. If you're on a Mac, try unzipping with The Unarchiver or Keka instead.

Ive never seen a rhyme or reason to this one and could never reproduce it here, so thats a good find! I dont have a mac at all here.

closed 2021-04-28 by InsanityAutomation

clue3030, 2021-04-28:
So how do we fix the sound from being broken. I have the same issue where the sound is broken and making strange sounds when I press something and the sound resets when I power back on. How do I fix this issue please.

InsanityAutomation, 2021-04-28:
The only things I have seen cause it were corrupted archives or poor SD cards. In every instance ive dealt with, or tinymachines reported from their facility, a fresh download extracted to a good SD card has resolved it. Locally ive only seen it occur from a corrupted archive, happened once when downloaded over a cellular connection. Tinymachines was working with a mac and saw it occasionally, but moving to a new SD card resolved it. In one case, the same SD card flashed a dozen machines, started corrupting sound on every one after that, new sd card flashed 100 more without an issue. Unfortunately thats about all of the guidance i can see on the issue outside of having one repeatably exhibiting it in front of me.

quarky42, 2021-04-28:
The only thing I have to add is a question: What utility are you using to extract the file?

Without being sure what operating system you're using: Can you try the latest version of 7-zip?  (That is what I've been using.)

clue3030, 2021-04-28:
I’m using a MacBook Pro 16”. I download the file and the Mac automatically unzips the file for me. I copy the diwin file to SD card and install it on the lcd

clue3030, 2021-04-28:
I also have a pc I could use (surface pro 7).

InsanityAutomation, 2021-04-28:
Sounds like the situation is exactly the same as what @JarrettWendt reported then, he reliably proved the Mac archiver was at fault in his case. The PC would be a safe alternative with 7zip.