InsanityAutomation/Marlin #118: Recommended firmware for stock Ender 5 plus?

pedromartinez11, 2020-04-10:
Which firmware file(s) are recommended for stock ender 5 plus ?
http://insanityautomation.com/Firmware/Creality/
I want to use bilenear bed leveling (if recommended)

is the stock Ender5 plus "dual z"?

I think i narrowed it down to one of these:
```
E5P_BILH_DW6.hex.gz      // hemera? cuz 'H' ?
E5P_BIL_DW6.hex.gz       // screen? 
E5P_BIL_DZH_DW6.hex.gz   // dual z? hemera, cuz 'H' ?
E5P_BIL_DZ_DW6.hex.gz      // dual z?
```

I looked at the abbreviations here https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

Jbchow, 2020-04-10:
I went through this over the last 2 weeks ago! Let see if I can help. 

That is the stable branch. I would start there If you have a stock Ender 5 plus with a V2.2 board you should choose either of these 2 options.

Bi-linear Bed Leveling:
`E5P_BIL_DW6.hex.gz`

OR

Unified Bed Leveling (What I chose)
`E5P_UBL_DW6.hex.gz`

Notes:
If you have V2.2 of Creality you do not have a silent board.

DZ: Both Z are on the same stepper driver so you want either. If you are on the V2.2 board the E1 (seconder extruder) stepper driver is not populated on the board.  Dual Z means it uses the E1 and the Z to drive each stepper separately. 

Update Screen Firmware: the Bleeding branch has a working screen config! `https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin2.0_Bleeding/CR-XABL_Screens_V2Rev2.7z `

Format sd card that came with printer to fat32 4k sectors (apparently it wont work if its not 4k sectors). copy the contents of the zip (DWIN_SET folder should be in the root of the sd card if this is done correctly) now remove the bottom cover from the printer plug in the sd card and wait till the top line says done. now remove the card and reboot the printer. Note: This version is designed for EXUI so it will not work with factory firmware.) 

To use UBL
UBL code:
Run to setup UBL: this will heat up bed and then probe bed for UBL then save to slot 1. I always level my bed fully before running this.

`M190 S60 ;Set Bed to 60c`
`G28 ; Home Printer`
`G29 P1 ;Runs Automatic Bed Leveling (UBL)`
`G29 P3 ;Extrapolates the leveling to parts the probe cannot reach.`
`G29 F10 ; Sets fade height to 10 mm`
`G29 S1 ; save bed topology to slot 1`
`G29 A ;ensures UBL is activated`
`@BEDLEVELVISUALIZER`
`G29 L1 ;Loads Mesh #1`
`M500 ;Saves settings to EEPROM `
`M140 S0 ;Turn-off bed`
`@BEDLEVELVISUALIZER`
`G29 T`

add to slicer:
In start g code find G28 and after that line add this

`G29 A ;activates UBL`
`G29 L1 ;loads UBL slot 1`
`G29 J; does a 3 point tilt correction on bed level`
`M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height`

boggisthecat, 2020-04-11:
Hi.

I have tried to install the update for the display, and it bombed out part-way through and now just displays red diagonal lines when powered up.

It won't attempt to install from the SD card any longer, so I assume the bootloader got borked.

Any ideas on how to reflash these?

Thanks.

> Update Screen Firmware: the Bleeding branch has a working screen config! `https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin2.0_Bleeding/CR-XABL_Screens_V2Rev2.7z `
>
> Format sd card that [...]

InsanityAutomation, 2020-04-11:
You would need to use the DGUS toolbox to flash over serial directly by plugging a PC adapter into the serial port on the display.

https://github.com/juliandroid/DWIN_CR_10s_Pro

I have not performed the procedure, as I have yet yo have a failure here and nobody has sent me a failed screen yet.

boggisthecat, 2020-04-12:
Hi.

Thanks for the pointer.  I can't find a PC with a hardware serial port, so some improvisation may be required.  If I can get this to work I will post up how.

Here is a photo of the display at present:

[image]

boggisthecat, 2020-04-12:
Hi again.

A fruitless few hours trying to reflash this.  I think the bootloader is borked.

This is the wiring:
[image]

It basically stays stuck, not acknowledging the boot command:
[image]
[image]

Also, the comms is doing a weird echo, with no power to the DWIN board, if you don't have GND connected:
[image]

I checked both the RX2/TX2 and RX4/TX4 pairs, and without the GND to serial it does this regardless.  This is why there is the weird double ground pin to the serial (it also doesn't want to share the ground if forked before the board pin).

The Targus breakout box I am using is pretty dodgy.  Some nice 118 Vac sitting on the 'grounded' surfaces.  Tingley.

boggisthecat, 2020-04-12:
Also, some info that may or may not be useful.

Unknown boot chip (presumed borked):
[image]

DWIN branded mystery chip:
[image]

My board had some interesting stuff on it:
[image]

boggisthecat, 2020-04-12:
Well... derp.

Probably shouldn't have looked at the wiring cables shown in the manufacturer's documents and assumed that they were wiring straight to a serial header.

Aside from the inverted logic, the voltage levels would be way out of spec for this device.  So it may not have been completely borked before, but probably is now.

(Retested everything with a TTL-RS232 converter this time.  Ugh.  Well, time to buy a new board.)

pedromartinez11, 2020-04-14:
> I would start there If you have a stock Ender 5 plus with a V2.2 board you should choose either of these 2 options.

My 5+ came with a 2.1 board. Would your steps still work?

[image]

Jbchow, 2020-04-14:
Should be good. the "silent board" uses different stepper drivers.
 `//#define CrealitySilentBoard // Creality board with TMC2208 Standalone drivers. Disables Linear Advance
The build listed in my post above runs the "A4988" stepper drivers (google tells me both 2.1 and 2.2 use these chips)

Also note this build has filament runout commented out (it comes from marlin head from what I can tell) Glad I tested it before I needed to use it. 

Clicking - That is normal. it only "Clicks" when it turns the probe into an idle state. UBL just leaves it out during all the probes.  - I printed a "bltouch shield" from thingaverse to protect the bltouch tip from crashes.

My config has it turned on. I am debating uploading my config to github. I tuned everything for the ender 5 plus I have. If you would find this useful let me know!

pedromartinez11, 2020-04-14:
@Jbchow 
>  I am debating uploading my config to github

Yea please! It would be very useful for me. 
Can you upload a gist of the git diff / patch?

Jbchow, 2020-04-15:
Here you go: https://github.com/Jbchow/Marlin/tree/CrealityDwin2.0_Bleeding-E5P-jbchow-Only
I had issues with the bleeding branch and printing off the sd card. I only wanted to print from the sd card to remove printing artifacts from buffer issues on octoprint. I found out about Melzi Host which disables  the sd card to allow for a larger buffer for octoprint.

pedromartinez11, 2020-04-16:
> Here you go: https://github.com/Jbchow/Marlin/tree/CrealityDwin2.0_Bleeding-E5P-jbchow-Only
> I had issues with the bleeding branch and printing off the sd card. I only wanted to print from the sd [...]

thank you @Jbchow ! Are you using the "bleeding" screen with that config? (https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin2.0_Bleeding/CR-XABL_Screens_V2Rev2.7z)

Jbchow, 2020-04-16:
@pedromartinez11 Yes! It works awesome.
I also made a wiki page with the steps from above: https://github.com/Jbchow/Marlin/wiki/Update-Screen-Firmware:-the-Bleeding-branch-has-a-working-screen-config
My Cura settings are in the read me section

pedromartinez11, 2020-04-18:
thank you @Jbchow it's working great! Closing this issue

closed 2020-04-18 by pedromartinez11