InsanityAutomation/Marlin #120: CrealityDwin_2.0 [DW6] - CR-10S Pro (possibly others) homing sequence - improving the life and accuracy of the switches

quarky42, 2020-04-13:
### Description

The Home X and Home Y routines should have a relative move of +10mm added to each of them. Currently, the homing sequence results in the print head pressed up against the X switch and the bed pressed against the Y switch. Maybe 10mm away from the switch is too much.  I think 5mm might not be quite enough to relieve the pressure from the switch.   So, maybe a minimum of 8mm to get them off the switches?

**On a side note:**
I think when a BLTouch is enabled the initial Z feedrate could be increased.  The second feedrate on the second test of the Z location can stay the same, but the initial "taste" of the z level could be done faster.  Maybe a 20% or 30% increase wouldn't be excessive?

### Steps to Reproduce
I'm not sure this is necessary in this case, but please let me know if it would help and I'll write it up.

**Expected behavior:** 

As in the description, I would like the Home X and Home Y routines to automatically move off of the switches after they have been homed.  

**Actual behavior:**

Currently, if you home X and Y, the limit switches are held down.  This can lead to some wear mainly in the spring behind the switch itself.  Limit switches are best when treated like momentary switches, clicked and then release the tension.   Keeping the springs and switches "tight" and "crisp" are a good thing.

#### Additional Information

I don't believe this is applicable, but if there is anything else you'd like me to provide, I'll do my best.

kisslorand, 2020-07-02:
I didn't like this also (endstop switches remaining pressed) so I made this change in MarlinCore.cpp:
```
    #ifdef EVENT_GCODE_SD_STOP
      queue.inject_P(PSTR(EVENT_GCODE_SD_STOP));
      queue.inject("G0 X10 Y10"); // by Lori
    #endif
```

kot0005, 2020-07-12:
how do I edit the marlincore.cpp file ? I have this issue too, I think its way too much brute force. I have a cr10s pro so I am using a hex file that tinymachines recommended.  its this one SingleExtruderScreens_V2Rev1.7z

InsanityAutomation, 2020-07-19:
Ive modified the HOMING_BACKOFF_POST_MM to 8mm for the next build. Should help the machines where 5mm isnt enough.

closed 2020-07-19 by InsanityAutomation

quarky42, 2020-07-19:
Thank you.  It didn't seem like there was any homing back off on my franken CR-10S Pro with SKR v1.4 turbo.  

It didn't seem like there was any homing back off when using the same firmware configured for the OEM silent board.  When turning the printer off the bed is firmly pressed against y limit, the x gantry is pressed against the x limit switch.... So your reply tells me there was supposed to be 5mm of back off that wasn't being applied when using either control board (silent or skr).

Should I wait for an incremental release to download and try it or should I download the GitHub zip and try it sooner?  I'm happy to give it a shot.

InsanityAutomation, 2020-07-20:
That and a few other tweaks are in my offline changes that'll get pushed to the bleeding branch tomorrow, you'll be able to give it a shot then!

kot0005, 2020-07-30:
> Thank you.  It didn't seem like there was any homing back off on my franken CR-10S Pro with SKR v1.4 turbo.
> It didn't seem like there was any homing back off when using the same firmware configured [...]

I am thinking of upgrading mine to SKR board too but first replace the stock heater and thermistor with slice engineering 24v ones and a mosquito, but will need to edit tables in the firmware

quarky42, 2020-07-30:
> I am thinking of upgrading mine to SKR board too but first replace the stock heater and thermistor with slice engineering 24v ones and a mosquito, but will need to edit tables in the firmware

No table editing needed.  That is my exact setup (well, I've got the Magnum).  There are already built in configs in this firmware, just set a couple flags in the firmware.

kot0005, 2020-07-30:
> I am thinking of upgrading mine to SKR board too but first replace the stock heater and thermistor with slice engineering 24v ones and a mosquito, but will need to edit tables in the firmware
>
> No [...]

Ohh that's Nice !! How would I make these changes though ? in the config file ? I will still be using stock motherboard for now.

quarky42, 2020-07-30:
I am on my phone right now, but I believe there is a Mosquito hotend flag.  I'd have to be looking at it to be sure.   The flag sets the proper thermistor value also.  (Just find the flag and then search through the file to see how that flag is used.)

kot0005, 2020-07-30:
> I am on my phone right now, but I believe there is a Mosquito hotend flag.  I'd have to be looking at it to be sure.   The flag sets the proper thermistor value also.  (Just find the flag and then search through the file to see how that flag is used.)

oh no worrries, take your time. I dont have the hotend yet. I have no idea how to set flags , am assuming there YT videos on how to do this but what are the keywords I should be using to search for a video guide ? also which file would I be editing ? All I did was upload hex file and lcd file to update my firmware. I know how to do basic coding in Arduino but this stuff is new to me.

can these flags be set using octoprint and sending commands or do I have to edit the file and upload the new firmware ?

kot0005, 2020-07-30:
also when you have time a link to this file would be awesome !

quarky42, 2020-07-30:
The flags are variables in Arduino.

Or even better look up how to use Microsoft Visual Code with PlatformIO extension added to it.  

In PlatformIO inside of Visual Code there are build flags that correlate to variables (configuration flags) in Configuration.h

You have to open the project INI file in PlatformIO.  I believe there is a guide linked in the InsanityAutomation GitHub here.  I know there are a couple of guides out there that will come up with some searching to help you get started.

kot0005, 2020-07-30:
> The flags are variables in Arduino.
> Or even better look up how to use Microsoft Visual Code with PlatformIO extension added to it.
> In PlatformIO inside of Visual Code there are build flags that [...]

I cant change these flags using Arduino IDE ?

kot0005, 2020-07-30:
https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Marlin/Configuration.h there seems to be a line to enable mosquito hotend.

quarky42, 2020-07-30:
> I cant change these flags using Arduino IDE ?

I did say "or".

So, yes you could use Arduino IDE, but once you learn how to use Platform IO by adding flags to the INI file instead of editing the Configuration.h directly, it becomes even easier.

The auto detect com port settings and flashing from inside Visual Code / PlatformIO works better.

There may already be a build setup for the Mosquito.  It would have a flag in the INI file 
-DVariableNane

So if the variable name was HotendMosquito, then you would just add -DHotendMosquito to the build flags.

The only thing I edit in Configuration.h now is just my probe offset, steps/mm, and my PID values so that the firmware has my correct values baked right in.  All the configuration flags are set at compile time and I can just copy those flags from one version of the firmware to a new version of the firmware that comes out.

quarky42, 2020-07-30:
> https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Marlin/Configuration.h there seems to be a line to enable mosquito hotend.
> 
> but it seems to be an unconfigured file :( So I would have to make a lot of changes.

What do you mean a lot of changes?

Again I encourage you to lookup how to use Visual Code and PlatformIO.  There might already be a premade configuration in PlatformIO that is close or even the same as your current configuration.