InsanityAutomation/Marlin #121: print finish button

bigbadbeau21, 2020-04-14:
i have the CR10spro running 10SPro_BIL_DW6.hex.gz file i am runnig Octoprint. at the end of the print my gcode is set to bring the build plate forward. Now there is a button that says finished printing i press this and it then says abort print then moves the y and x to end stops. can you delete this button and just make it go to home screen or stop it from moving after pressing the finished button.

KillSwitch422, 2020-04-14:
I have the same action, is it possible we are missing a new command in the ending script Gcode? I was running B7 previously .

myrandor, 2020-04-18:
Same thing with CR-10S Pro using compiled firmware.

Hyrules, 2020-04-24:
Same here. It don't know if it's a feature or a bug.

quarky42, 2020-04-29:
With my CR-10S Pro and no Octoprint, the print is presented with Y300 automatically at the end of the print.  When I press the Finish button the printer homes X and Y.

The homing sequence leaves the X and Y parked on the limit switches.   This is also not good behavior.  Limit switches should be momentary contact, not sit and park on them.

PHIL-IP, 2020-06-13:
Same as previous, if I clic the button named "Finish print" then the printer display "Print aborted" .... !!!
In my G-Code I disable steppers at the end of the print but at the moment the head is parcked at home and steppers are not disabled :(

nibblerrick, 2020-10-15:
Good to know that I am not alone with that. Same problem.

InsanityAutomation, 2021-04-28:
Ive set the complete action to simply be M84 for the next builds, disable steppers. Originally it was set to home XY as many were reporting they didnt want the head left hot over the part, but its probably safest to just disable steppers and cooldown where its at.

closed 2021-04-28 by InsanityAutomation

quarky42, 2021-04-28:
> 
> 
> Ive set the complete action to simply be M84 for the next builds, disable steppers. Originally it was set to home XY as many were reporting they didnt want the head left hot over the part, but its probably safest to just disable steppers and cooldown where its at.

Would you consider this pseudo code approach instead:

If (current_z + 15mm < current_machine_max_z)
then
raise the z position up by 15mm
else
raise the z position up to current_machine_max_z

15mm in that pseudocode is arbitrary and could be whatever z move you think is best.  Just moving away from the part would be really good.

Another thing to consider adding right before that pseudocode is to turn the cooling fan on to at least 50 percent.  Assuming that the the hot end is off at that point or will be turned off by the firmware momentarily, and you add a timer to turn off the cooling fan after 1 minute, that nozzle will have cooled down tremendously, and the cooling fan only needs to run a little while to rapidly cool the nozzle.   (30 seconds of that might be enough.)

InsanityAutomation, 2021-04-28:
Soft limits could actually control the raise, so I can simply make the abort script : G91;G1Z15;G92;M84M104S0;M140S0;

Im not sure cooling fan is really good to force in abort, if the print is aborted because its spaghetti everywhere dont want to kick the fan on and grab some. It will cool it faster but not sure thats needed

quarky42, 2021-04-28:
I would exclude abort print from that fan idea for the same reason, but if you want the same behavior regardless of abort or print complete, yeah I can see using the soft limits.

I'm leary of soft limits because I've seen crashes, but it has been long enough ago that I can't recall the circumstances of the crashes.

Recently, I have seen the X soft limits working, so I really don't have a strong reason to be leary of them, I just don't like stripping belts or stalling motors against lead screws.

Is the math check prohibitive considering the print is finished?

InsanityAutomation, 2021-04-28:
Its not prohibitive, it just puts the whole command in ram instead of progmem. Trying to reduce ram consumption so I can expand buffers for serial printing as much as possible.

I havnt seen an issue with soft endstops in quite awhile, but id think the possibility of an aborted print being that close to the top is low. 

At that point on the screen, it doesnt know if it was aborted or finished, the screen actually sends the same command from the stop button as the finish button... Now that weve made the jump away from interoperability maybe this can be improved but its much lower on the priority list. Ill probably work on standby brightness and storing volume in eeprom next. Maybe for now ill set it to raise 5mm.

quarky42, 2021-04-28:
Thank you.  Sounds great and I really appreciate the education here!