InsanityAutomation/Marlin #122: Filament runnout sensor triggering at strange times, and causing system freeze CR10SPro

KillSwitch422, 2020-04-14:
### Description

Filament Sensor issue in CR10s pro, running 10SPro_BIL_DW6
### Steps to Reproduce

Experienced the same issue on two separate occasions.
1. Run Bed leveling from LCD, Wait for home, select measure, status heating Bed, probes all location fine, then Heating E, during this step the no filament screen comes up: do you want to change filament. If yes is selected the printer hangs, and needs hard reset.
2. run M303 S200 for PID from proterface, get does it successfully and gives results in console, then the lcd gives gives no filament screen (even when there is filament loaded).

If I Run the M600 command it executes fine, with correct lcd selections and buttons.

bigbadbeau21, 2020-04-14:
i have the same and never got the no filament screen, putt a bit of filament in and see if it does it again.

KillSwitch422, 2020-04-15:
even when filament is loaded if I run M303 same thing

KillSwitch422, 2020-04-17:
Apologies this is partly covered by #90 , I am also sending the commands via Proterface. 
Why it is doing the filament check, after these actions is still a question in my opinion.

InsanityAutomation, 2020-07-19:
Ive heard of this coming from pronterface but have yet to see a log from pronterface of when this occurs. I can track it down further if I can get the log.

InsanityAutomation, 2021-04-28:
The latest updates did make some changes to the PID tune calls to help avoid this as well as proper messaging around it.
Upstream currently stops sending data to the LCD during PID tuning as the extui idle function is not called, so there is a little more room for improvement, but one step at a time...

Regardless, the issue here appears to match the odd injections of M0 by pronterface.

closed 2021-04-28 by InsanityAutomation