InsanityAutomation/Marlin #123: Auto leveling at the beginning of printing

mateo356, 2020-04-15:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

masc2279, 2020-04-20:
There is a much simpler way Just add a G29 to your start script and it all works perfect ....

masc2279, 2020-04-20:
I tested this also on my machine and it does not work CR-10 MAX. But as i posted above just use a G29 which is a work around.

mateo356, 2020-04-21:
Yes, I putted G29 and M500 at the beginning of G code, without M500 just done auto bed leveling without saving those value.

elecktron3d, 2020-06-03:
i have those two gcodes on the Start GCode Section.

G28 ;Home
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

Do you recommend to change those gcode to:
G29
M500

mateo356, 2020-06-03:
Hi, 
I'm using Cura for generating g-code, and If you want auto bed leveling at the beginning the 3d print you need to put two those g-code line  after G28 and before M420 S1 Z2, look image in attachment.
[image]

elecktron3d, 2020-06-03:
@mateo356 thank a lot! ill try it

elik745i, 2020-12-14:
I'm using Creality CR-10 MAX and I can't disable auto bed leveling through menu, does not work DW6.2 not  in DW7 why? Everything was working on original CR-10 MAX firmware...

InsanityAutomation, 2021-04-28:
The standard process is to recall the mesh from the EEPROM at the start of each print rather than reprobing. Reprobing constantly brings in a large margin of error. Its considered best practice to probe once and store a good validated mesh and restore it. It is currently configured to enable leveling if there is a valid mesh on startup.

closed 2021-04-28 by InsanityAutomation