InsanityAutomation/Marlin #125: [CrealityDwin_2.0][DW6] AllMetalHotEnd defaulted to on in Configuration.h ?

quarky42, 2020-04-16:
### Description

Starting at line 50 of Configuration.h:

```
//Enable this if you have an all metal hotend capable of 300c
#ifndef HotendAllMetal
  #define HotendAllMetal
#endif
```

If HotendAllMetal is not defined, then define HotendAllMetal.   Doesn't this block automatically define HotendAllMetal for all configurations no matter what since it is uncommented by default?

Should the HotendAllMetal really be defaulted on like this?  Should this block be commented out until someone actively decides they need to turn it on?

### Steps to Reproduce

I've just had this question after several times of setting up this firmware (twice on one printer and once on a new printer.  The first printer has an all metal hotend and the second one will get one pretty soon, so I don't mind having it on in my situation, but wondering if others are having this flag turned on by accident?

**Expected behavior:** [What you expect to happen]

I thought this flag would be off by default.

KillSwitch422, 2020-04-17:
The #ifndef command will only do this if there is no definition present prior , thus if you don't select one of the options it will default to all metal hotend AFAIK

> //#define HotendStock
//#define HotendE3D
//#define HotendMosquito

quarky42, 2020-04-17:
Sure, if it said:

```
#if !defined (HotendStock) && !defined (HotendE3D) && !defined (HotendMosquito)
    #define HotendAllMetal
#endif
```

then I would agree with you and it would be doing what you are saying it is doing.

What it actually says is just:
```
#ifndef HotendAllMetal
    #define HotendAllMetal
#endif
```

and all that means is if HotendAllMetal isn't defined, then define it which is completed every time no matter the status of the other flags because the if statement isn't actually tied to any of those flags as far as I can see.  So, it isn't doing what you said, and my original question still stands.    Is what is currently there really correct?   The comment in the code and your reply makes it seem like it should be checking to see if none of the hotend variables are set and then setting the HotendAllMetal, OR it should be commented out and someone who needs it should be uncommenting it because of the way it is currently written.

I think it should be something like what I have in the top of this reply.  I have written in c/c++ and a number of other languages, so I know I could be missing something, but I can't see it at that point.  It seems pretty straight forward that the psuedocode should read like:

```
if these other variables aren't defined
    then define this HotendAllMetal
end if
```

I hope this makes sense and if I am missing something here you can explain it to me.  Thank you very much for your time and discussion of this.

because

KillSwitch422, 2020-04-17:
on second look, I think you right it doesn't appear correct, it would work if there is common def , but my c is really rusty, so I possibly shouldn't be second guessing..

I understand the the indef will check if it is defined previously and if not apply the condition
this could work, but it should work the same as your code

```
//#define Hotend HotendStock
//#define Hotend HotendE3D
//#define Hotend HotendMosquito

#ifndef Hotend
  #define Hotend HotendAllMetal
#endif
```
would result in HotendAllMetal

```
#define Hotend HotendStock
//#define Hotend HotendE3D
//#define Hotend HotendMosquito

#ifndef Hotend
  #define Hotend HotendAllMetal
#endif
```
would result in HotendStock etc

.

quarky42, 2020-04-17:
Yeah, now I think you're seeing what I'm saying.  I agree, something like that would work the way I think was intended in that section.   I've looked at that section off and on for a long time (since some time last year) and it just never sat right with me.  I finally decided to say something.

I like what you came up with.  That is likely the simplest interpretation of what I think they meant to do there.  Thank you again for taking the time to discuss this.   I can test that syntax out next time I go to update my board (which will likely be middle of next week at the latest) and submit it as a fix.

Personally, if the dev could take a quick look at this and give us their take on the existing block of code vs the one you propose and see if they can see the same disconnect I was concerned about, that would help me feel better.  I don't like editing other people's code unless I'm 100% sure I understand what is going on and I'm only at around 95% at this point.   I was at 85% until you came along and lent another pair of eyes to this.  So, sincerely, thank you again for adding to this discussion!

jxlarrea, 2020-04-24:
It's definitely a typo. As it is at the moment `HotendAllMetal ` is always defined. I ended up commented that line and defining `HotendStock` in the build arguments.

quarky42, 2020-04-24:
I noticed line 373:

```
#if NONE(HotendStock, HotendE3D)
  #define HotendStock
#endif
```

Which should probably read:

```
#if NONE(HotendStock, HotendE3D, HotendAllMetal)
  #define HotendStock
#endif
```

And the block near the top could either be removed or just simplified to a single line added to:

```
//#define Hotend HotendStock
//#define Hotend HotendE3D
//#define Hotend HotendMosquito
//#define Hotend HotendAllMetal
```

I believe that brings the HotendAllMetal into line with everything else taken into account (looking at all references to Hotend through the file.)

InsanityAutomation, 2020-05-05:
Yeah, when configuring everything with the autobuilds in platformio, we didnt want to double the number of hex files. We determined for now we will just default it on for everything and if there is not a large number of requests to limit the temperatures to the PTFE ranges in select builds, we will just keep it that way and eliminate the option.

quarky42, 2020-06-03:
What about my last message?   Looks like around line 421, currently:
#if NONE(HotendStock, HotendE3D)
  #define HotendStock
#endif

Should it not be:
#if NONE(HotendStock, HotendE3D, HotendMosquito, HotendAllMetal)
  #define HotendStock
#endif

That way if I define my HotendMosquito, then I don't end up getting a HotendStock being defined as well?

This isn't about changing the builds, but rather making the logic consistent.   My understanding of the code isn't thorough, so I may be missing something that is part of this.

InsanityAutomation, 2020-07-19:
All metal just sets temp range, so that one is irrelevant there as its was intended to coincide. Mosquito is irrelevant there as well as it also only changes PID and thermistor, not probe offsets as hotend stock is enforcing.

closed 2020-07-19 by InsanityAutomation