InsanityAutomation/Marlin #128: Ender 5 Plus bed size in autobuilt hex file

jxlarrea, 2020-04-23:
A couple of days ago I installed the firmware for my Ender 5 Plus (with silent board) located at [](https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Hex%20Files/E5P_UBL_Slnt_DW6.hex.gz). 

I noticed that the bed size is specified as 360x360 instead of 350x350. Is this a mistake or is there a particular reason for this discrepancy?

jxlarrea, 2020-04-24:
It seems like having the bed size larger than the physical bed causes the probe to crash against the frame by a few millimetres (~5mm) when the X position is set to 0 using the Move menu on the touchscreen.

It also causes the probe to crash really hard against the frame when using the "validate mesh" button on the Aux leveling screen.

jxlarrea, 2020-04-24:
Is there any value I can modify on the Configuration.h file to avoid the probe crashing against the frame?

cuboy29, 2020-04-24:
try to relocate the x end switch.

Uforias, 2020-04-24:
This link will help you changing the values on configuration.h: https://reprap.org/wiki/Configuring_Marlin_Bed_Dimensions

jxlarrea, 2020-04-24:
I ended up changing the bed dimensions in the Configuration.h file. I'm just wondering why is the Ender 5 Plus bed size defined as 360x360 instead of 350x350 on this firmware. Is it a typo or is there any purpose to it? Something to do with UBL perhaps?

cuboy29, 2020-04-24:
i noticed the size is also 360 in the source that Creality just released for the printer as well.

bpivk, 2020-05-23:
Because the actual bed size is in fact 370x360.

closed 2023-02-17 by InsanityAutomation