InsanityAutomation/Marlin #129: Hot end Heating after Bed leveling complete

Hyrules, 2020-04-24:
### Description

After doing a bed leveling the hotend heats and place itself in a position which is not home. I'm not sure if this is a feature or a bug. I'm using DW6 and i'm on a CR10S-PRO

### Steps to Reproduce

1. Heat the bed to 60 degrees
2. Wait a minute or 2 for the bed to heat up.
3. Press the measure button

**Expected behavior:** [What you expect to happen]

Bed level finishes and the printer home itself hot end does not heat up.

**Actual behavior:** [What actually happens]

Bed level finishes and the printer heats the hotend and place itself at a point in the bed.

#### Additional Information

[Config.zip](https://github.com/InsanityAutomation/Marlin/files/4530809/Config.zip)

I'm still on the bleeding Edge version that happen to fix the out of filament error after the print DW6.1

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jxlarrea, 2020-04-25:
I noticed this as well. The "Measure" routine actually heats up the bed automatically first up to 65C. Then it does the ABL measurement. Once it finishes, it heats up the nozzle and asks to "Set Z-Offset". Nothing happens afterwards and the hot-end remains heated.

toniino38, 2020-04-29:
Same issue for me, it heats up the nozzle to 215C and asks to "Set Z-Offset" crushing the nozzle on the bed.

InsanityAutomation, 2020-05-10:
This is part of the standard commissioning script used across many machines for years. This would go to wherever the current z offset is defined, meaning Z0. If the machine can home and go to 0 correctly, which it should before doing the bed leveling routine, then the final step of fine tuning the z offset should never crash. Home and set 0 first, then run measuring. You can check the actual z offset value in the tools screen and raise it if needed beforehand.

closed 2020-05-10 by InsanityAutomation

Hyrules, 2020-05-10:
So you are telling me that at the end of a measuring sequence the hot end is supposed to heat up right ? I don't have any problem with the nozzle crashing in the bed but why should the hotend heat up after the measuring sequence is finished ?

MorganTr44, 2020-05-11:
HI,
Same problem : (END OF MEASURING)
leveling screen selection :

1: "homing" for  Z offset adjust        : OK
2: Select measuring
  - "homing"                                         : OK
   - bed rise 55°c                                 : OK
   - levelling on 25 points                    : OK 
   - homing                                            : OK (msg: Bed levelling done)
     can't quit process at this time
   - **hotend  rise 215 °**                           : **?**  (msg: E heating)
   - **hotend go to x:100 Y:110 Z :0**       : **?**
     then the filament comes out under the effect of heat and damage my bed (ziflex).
   - **Nozzle and bed temp stay at 215/55**  : **?**    

To avoid this I have to wait for the last displacement and quickly move Z manually.
PS: (CR-10s PRO V2)

LMS0815, 2020-05-14:
> So you are telling me that at the end of a measuring sequence the hot end is supposed to heat up right ? I know that the nozzle crashing in the bed is not normal but it's not an issue i'm having. Why should the hotend heat up after the measuring sequence is finished ? Heating the bed and hotend before is one thing after is another.

It is expected to finetune Z offset as the last step of bed leveling. Therefore the hotend is heated for better accuracy. Z offset is moved to the **old value**. If it is not 0 it might crash bed. _(Consider setting Z offset prior to bed leveling by M851 Z0)_ If you move Z and leave the menu bed and nozzle temperatures are switched off immediately If not this will happen after some timeout.