InsanityAutomation/Marlin #130: Unable to compile

jimmyeao, 2020-04-26:
### Description

Unable to compile

### Steps to Reproduce

Edit configuration.h for stocker ender 5 plus and bltouch
Compile
**Expected behavior:** [What you expect to happen]
FW compiles

**Actual behavior:** [What actually happens]
hundreds of errors and it seems to be compiling for every option possible

#### Additional Information
[Configuration_adv.zip](https://github.com/InsanityAutomation/Marlin/files/4535312/Configuration_adv.zip)

I cant find any documentation on how to correctly configure this package for a stock Ender 5 plus (or any other machine)
Im not new to vscode, or Marlin, but I seem to be missing something here.

My goal is to update the stock board so I can use M500 (and linear advance if space), and to then compile for an SKR1.3 and TMC2209 which I have coming.

Any pointers gratefully received :)

cuboy29, 2020-04-26:
Don't edit the configuration.h

Compile using these profiles.  I just edit platformio.ini instead.

[image]

jimmyeao, 2020-04-26:
so I just delete anything I don't want under default_envs=  (Leaving E5P_BIL)?

cuboy29, 2020-04-26:
you don't have to delete anything.  just click on build button for the E5P_BIL only and it'll just compile for that version.

jimmyeao, 2020-04-26:
got it, thanks. Not seen it used like that before :)

closed 2020-04-26 by jimmyeao