InsanityAutomation/Marlin #131: Error when compiling for SKR1.3

jimmyeao, 2020-04-27:
### Description

Error compiling for SKR1.3

### Steps to Reproduce

1. edit platform.ini for Ender 5 plus:
```
[env:E5P_BIL]
platform = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip
board = nxp_lpc1768
board_build.f_cpu = 16000000L
lib_deps = ${common.lib_deps}
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = ${common.build_flags} -DMachineEnder5Plus -DHotendAllMetal -DBedDC -DPOWER_LOSS_RECOVERY -DForce10SProDisplay
src_filter = ${common.default_src_filter} +<src/HAL/AVR>
```

2. Edit configuration.h:

```
#define SKR13 // 32 bit board - assumes 2208 drivers
//#define SKR14
//#define SKR14Turbo
//#define SKRPRO11

//#define I2C_EEPROM  // use I2C EEPROM on SRK PRO v1.1 e.g AT24C256
//#define SKR14_PowerLossKit // Bigtreetech power loss kit for SKR14

#define SKR_2209
#define SKR_UART // Configure SKR board with drivers in UART mode
//#define SKR13_ReverseSteppers // Some users reported directions backwards than others on SKR with various drivers.
#define DualZ // Uses 5th driver on CRX or SKR boards as Z2
```
3. Compile
**Expected behavior:** [What you expect to happen]
Firmware compiles

**Actual behavior:** [What actually happens]
Errors:
```
In file included from Marlin\src\feature\../module/stepper/trinamic.h:29,
                 from Marlin\src\feature\../module/stepper/indirection.h:44,
                 from Marlin\src\feature\../module/stepper.h:47,
                 from Marlin\src\feature\babystep.cpp:30:
C:\users\jimmy\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:14:10: fatal error: SPI.h: No such file or directory
```
#### Additional Information

Perhaps I am setting up the firmware incorrectly?
Goal is to compile for an SKR1.3 with TMC2209s in UART fro an Ender 5 Plus.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jimmyeao, 2020-04-27:
ok, figured it, was doing it totally wrong! Sorted now

closed 2020-04-27 by jimmyeao

davidsteinberger, 2020-05-12:
@jimmyeao I'm having the same problem (probably same mistake) compiling to the skr 1.4 turbo. Could you quickly describe how you got your build working on the LPC176X?