InsanityAutomation/Marlin #132: Ender 5 Plus Display on SKR1.3?

jimmyeao, 2020-04-28:
### Description

I have managed to flash the screen with the CR-XABL_Screens_V2Rev2.7z firmware, and have the SKR1.3 working in the Ender 5 Plus, unsure how (or if) I can connect the original display to the SKR1.3?

jimmyeao, 2020-04-29:
update. gut the lug off the connector and reversing the plug now brings the screen to life, but I have this 
[image]
Octoprint works fine

cuboy29, 2020-04-30:
can you share you config files?  i am thinking of changing out my board to this too.  Might go with the TFT3.5 for now.

jimmyeao, 2020-05-01:
https://deviousweb-my.sharepoint.com/:u:/p/jimmy/ERv4dS15uGlHon-RgYecZ68BULj_-5eNfssLpjIY7Wgp6g?e=V478fQ
Bear in mind, this contains the following changes to make it work with the E5+
[image]

cuboy29, 2020-05-04:
Thanks.. I want to try and install the SKR1.3 E3 version.

InsanityAutomation, 2020-05-05:
If no data is getting to the screen, try swapping the TX/RX pins. Remember mobo TX goes to screen RX and so on.

jimmyeao, 2020-05-05:
Success! Made my own cable and connected to the TFT header on the SKR, now all working.

closed 2020-05-05 by jimmyeao