InsanityAutomation/Marlin #134: Quality issue with DW6 2.0.5

daredevil-ferreira, 2020-05-04:
### Description
I updated both motherboard & screen firmware.
And try to print these parts :
             https://drive.google.com/open?id=14RO5IvE9s4Z2HT-RN2ahGumINT-ekl8X
[image]

with 2 different filaments.

I have a CR10s Pro and launch these prints via Octoprint.

The round shapes are ugly with a lot of blobs where as the length are more or less ok.

I noticed that the small movement are not continuous as if the motherboard was not able to execute smoothly the gcode.

Is it the Octoprint interface of the fact than now the screen displays every movement (X, Y, Z) while printing ?

I am going to test it directly from an SDcard.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

chunter1, 2020-05-04:
> I noticed that the small movement are not continuous as if the motherboard was not able to execute smoothly the gcode.

I observed the same after updating a few days agoe on my CR-10S Pro.
However, i can't tell if the new firmware is less performant in executing micro movements.
Maybe temporarely disabling the LCD interface feature to check any performance issues could help?
Can you provide the link or gcode of your model for further investigations?

daredevil-ferreira, 2020-05-04:
My GCode 
https://drive.google.com/open?id=10i5Plwj045Gf7a5AVMZxOFtqspzT7BaU

[CCR10SPRO_lid (1).gcode.txt](https://github.com/InsanityAutomation/Marlin/files/4574072/CCR10SPRO_lid.1.gcode.txt)

How can I disable the LCD ?

chunter1, 2020-05-04:
> How can I disable the LCD ?

Don't know exactly how but maybe by disabling this:

`//
// Third-party or vendor-customized controller interfaces.
// Sources should be installed in 'src/lcd/extensible_ui'.
//
#if ANY(MachineCR10SPro, MachineCRX, MachineEnder5Plus, MachineCR10Max) && (DISABLED(GraphicLCD) || ANY(Force10SProDisplay, ForceCRXDisplay))
  #define EXTENSIBLE_UI
#endif`

FrancescoTalotta, 2020-05-05:
Hello. I noticed the same. Flashed the firmware 5 days ago and I have a lot of bubbles in the curved surfaces. I have a CR10s Pro with Octoprint. 

[image]

riri19, 2020-05-05:
Same for me :(
[image]

daredevil-ferreira, 2020-05-05:
I had the same conclusion between SD & Octoprint
SD:
[image]

I did suspect an issue with the baud rate and I swith my raspberry 3B to a raspberry 4B and things get a bit better but still not as good as the SD print.
Raspberry 4 vs Raspberry 3:
[image]

Raspberry 4 vs SD:
[image]

I am printing form the SD with same filament to verify.
If there is still a reference, I will try to change the firmware for 115000 bauds.

riri19, 2020-05-05:
I hope you find a solution :)

chunter1, 2020-05-05:
Did you compile the firmware yourself and did you use the provided example config file for the  CR10S Pro?

beny18241, 2020-05-05:
same here, using provided hex file.

FrancescoTalotta, 2020-05-05:
I compiled the firmware with platformio, using the provided configuration for the CR10S Pro

chunter1, 2020-05-05:
Comparing the default "Configuration_adv" to the cr10spro specific one, i see the following difference in the serial buffer config:

**DEFAULT:**
// The ASCII buffer for serial input
...
#if ENABLED(MachineCR10Orig) //melzi has more ram than a 2560
  #define BUFSIZE 16
#elif ANY(SKR13, SKR14, SKR14Turbo, SKRPRO11)
  #define BUFSIZE 8
#else
  #define BUFSIZE 4
#endif

**CR10SPRO:**
#define BUFSIZE 4

daredevil-ferreira, 2020-05-05:
I did use the hex provided.
The yellow filament turns ok when printed from the SD.

I just had a look to the  1.60.3 firmware provided by creality, the BUFSIZE is indeed set to 4.

As I understand the source code firmware, for the CR10sPro:
   - The environment compiles the hex file with -DMachineCR10SPro
  - The configuration_adv.h in this case set the BUFSIZE to 4 also (we are not in the MachineCR10Orig nor SKR13, 14, 14Turbo, PRO11)

chunter1, 2020-05-05:
> * The configuration_adv.h in this case set the BUFSIZE to 4 also (we are not in the MachineCR10Orig nor SKR13, 14, 14Turbo, PRO11)

We are in the "MachineCR10Orig"

daredevil-ferreira, 2020-05-05:
In my configuration.h I have 
//#define MachineCR10Orig // Forces Melzi board

Later in the file 
#if ANY(MachineEnder2, MachineEnder3, MachineEnder5, MachineCR10, MachineMini) && NONE(Melzi_To_SBoardUpgrade, SKR13, SKRPRO11)
  #define MachineCR10Orig
#endif

but all: 
//#define MachineEnder2
//#define MachineEnder3
//#define MachineEnder5
//#define MachineMini
//#define MachineCR10

are also in comments.

I see that 
#if ENABLED(MachineCR10SPro)
  #define MachineCR10Std

But no MachineCR10Orig

daredevil-ferreira, 2020-05-06:
The firmware change at baud rate set to 115000 did not change quality.
I am testing with BUFSIZE set to 4

chunter1, 2020-05-06:
> I see that
> #if ENABLED(MachineCR10SPro)
> #define MachineCR10Std
> 
> But no MachineCR10Orig

Sorry, you are right.
I confused MachineCR10Orig with MachineCR10Std.

daredevil-ferreira, 2020-05-06:
No difference in quality of the print despite changing baud rate and BUFSIZE

I have no clue for any new testing.

daredevil-ferreira, 2020-05-08:
Many several tries, last : BUFSIZE = 8 , BLOCK_BUFFER_SIZE = 16, baud rate back to 250000

Nothing works. I am going back to the 1.70 version

chunter1, 2020-05-23:
Anyone found the reason or a fix for that issue yet?
I just printed some stringing towers and there is definitive a huge problem when it comes to printing circles (=a lot of short motions in short short time).

daredevil-ferreira, 2020-05-23:
No, and I found the same issue with the version 1.70.
I was not able to find the first version of the firmware provided by Tiny Machines (I think they called it B6) to verify if the USB issue with circles was there.
I did print a lot of items though with this first Tiny Machines firmware and I had never noticed this issue.

chunter1, 2020-05-23:
Since the Marlin 2 Version, the coordinates of the extruder are shown in realtime on the display.
Could it be that the permanent serial communication to the display to update the values introduces a bottleneck?

daredevil-ferreira, 2020-05-24:
where can I get the 1.1.9_B7 files ?

daredevil-ferreira, 2020-05-24:
I will try to comment the 
    SERIAL_ECHOPAIR_P(X_LBL, lpos.x, SP_Y_LBL, lpos.y, SP_Z_LBL, lpos.z, SP_E_LBL, lpos.e);
in the report_logical_position of the motion.cpp file to see if this has an impact.

riri19, 2020-05-24:
Good idea

chunter1, 2020-05-25:
> where can I get the 1.1.9_B7 files ?

It's under the "Creality_DWINTest" branch:
https://github.com/InsanityAutomation/Marlin/tree/Creality_DWINTest

chunter1, 2020-05-26:
> I will try to comment the
> SERIAL_ECHOPAIR_P(X_LBL, lpos.x, SP_Y_LBL, lpos.y, SP_Z_LBL, lpos.z, SP_E_LBL, lpos.e);
> in the report_logical_position of the motion.cpp file to see if this has an impact.

Did you try it?

daredevil-ferreira, 2020-05-27:
Hi not yet, I have a large print still on going. I will try this weekend.

Regards

chunter1, 2020-05-29:
Unfortunately commenting out "SERIAL_ECHOPAIR_P(..." does not make a difference.

chunter1, 2020-05-29:
I discovered another problem while printing my default stringing towers.
The print attached has 3 outer perimeter and should be filled.
However if you look at the picture, you see the top layer filling offset!
It is shifted.
[image]

chunter1, 2020-05-30:
I just adapted the latest original Marlin version "bugfix-2.0.x" for the CR10S-Pro without display support.
Hard to say, but i think the print is a bit better - however, i suspect the issue generally started with the transition to Marlin 2.0.
Can it be that the focus on more performant 32-bit platforms leaves the 8-bit ones running into trouble?

mihaiserban, 2020-05-30:
same issue for me. upgraded a stock ender 5 plus to DW 6.2.
I got these artifacts while printing round objects. I need to slow it down significantly to avoid them.
[image]

I noticed if i speed it up, there's a weird sound coming from the motors. Posted a video on reddit.
https://www.reddit.com/r/ender5plus/comments/gnw43r/tinymachines_firmware_62_stock_ender_5_plus_weird/

daredevil-ferreira, 2020-05-30:
I have the same issue with the 1.1.9 B7.

I've asked tiny machine where could I get the initial firmware (B6) I had.

jxlarrea, 2020-05-30:
Confirmed. I'm also getting weird noise like artifacts when printing round objects using the TinyMachines firmware on my Ender 5 Plus. Printing from OctoPrint.

[image]

chunter1, 2020-05-30:
> Hello. I noticed the same. Flashed the firmware 5 days ago and I have a lot of bubbles in the curved surfaces. I have a CR10s Pro with Octoprint. Do you know a commit that I can checkout in order to get rid of this problem?
> 
> [image]

Looks like the bandwidth issue of the UART communication as explained well in this video:
https://www.youtube.com/watch?v=Hvw3DrVAeTA

mihaiserban, 2020-05-30:
Kersey Fabrications has been running in similar quality issues with Marlin 2.0 - https://www.youtube.com/watch?v=smQ3JX_P0vE 
I'll try to disable `//#define JD_USE_LOOKUP_TABLE` as suggested in this [Marlin 2.0 bug](https://github.com/MarlinFirmware/Marlin/issues/17920)

chunter1, 2020-05-30:
> Kersey Fabrications has been running in similar quality issues with Marlin 2.0 - https://www.youtube.com/watch?v=smQ3JX_P0vE
> I'll try to disable `//#define JD_USE_LOOKUP_TABLE` as suggested in this [Marlin 2.0 bug](https://github.com/MarlinFirmware/Marlin/issues/17920)

Isn't this only relevant when junction deviation is enabled which i don't have activated but using classic jerk.

`#if HAS_JUNCTION_DEVIATION
  // Enable this option for perfect accuracy but maximum
  // computation. Should be fine on ARM processors.
  //#define JD_USE_MATH_ACOS

  // Disable this option to save 120 bytes of PROGMEM,
  // but incur increased computation and a reduction
  // in accuracy.
  #define JD_USE_LOOKUP_TABLE
#endif`

mihaiserban, 2020-05-30:
> Isn't this only relevant when junction deviation is enabled which i don't have activated but am using classic jerk instead.

indeed.. JUNCTION_DEVIATION is disabled by default in TM firmware right?

chunter1, 2020-05-30:
> > Isn't this only relevant when junction deviation is enabled which i don't have activated but am using classic jerk instead.
> 
> indeed.. JUNCTION_DEVIATION is disabled by default in TM firmware right?

Yes, disabled by default.

daredevil-ferreira, 2020-05-31:
Anyway how this parameter can have a distinct impact regarding if the print is sent via Octoprint or the SD card ?

chunter1, 2020-06-01:
Has anyone got the Marlin 1.1.6 Version of Tiny Machines?
As mentioned in the video below, the quality should be fine with it.
https://www.youtube.com/watch?v=smQ3JX_P0vE

chunter1, 2020-06-02:
I'm switching to Klipper firmware since i wanna get rid of that annoying Creality DWIN display solution on my CR10S Pro anyway.

daredevil-ferreira, 2020-06-04:
What config did you use in Klipper ?
I thought that Klipper was not running on the CR10s Pro (https://github.com/KevinOConnor/klipper/issues/1113)

Regards

chunter1, 2020-06-04:
> What config did you use in Klipper ?
> I thought that Klipper was not running on the CR10s Pro

That issue is already fixed.
I have setup my own Klipper config by using the example config in the wiki.

chunter1, 2020-06-04:
That's awsome!...
With OctoKlipper you get performance graphs of your Klipper system :)

[image]

daredevil-ferreira, 2020-06-21:
I am using Klipper and I am also very pleased.
Bed Mesh leveling works even better than with marlin

Hoffi-D3, 2020-06-22:
> > What config did you use in Klipper ?
> 
> > I thought that Klipper was not running on the CR10s Pro
> 
> 
> 
> That issue is already fixed.
> 
> I have setup my own Klipper config by using the example config in the wiki.
> 
> 

Hi

I have the same issue.

Could you please share your config with me?
I want to give klipper a try.
Thank you so much

chunter1, 2020-06-22:
Here is my Klipper config for the CR10S Pro so far. 
Please note, that I modded the TMC2209 on the Creality motherboard to support UART.
That way, i can use pressure advance (linear advance) on my CR10S Pro. :-)

`# This file contains pin mappings for the 2017 Creality CR-10S. To use
# this config, the firmware should be compiled for the AVR atmega2560.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 310
homing_speed: 80

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 310
homing_speed: 80

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe: z_virtual_endstop
position_min: -5.0
position_max: 400

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .00717641
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
min_temp: 0
max_temp: 250
min_extrude_temp: 170
max_extrude_only_distance: 600
pressure_advance: 0.5
max_extrude_only_velocity: 200
max_extrude_only_accel: 3000

[tmc2208 extruder]
uart_pin: ar33
microsteps: 16
interpolate: True
run_current: 0.730
sense_resistor: 0.150

[heater_bed]
heater_pin: ar8
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
min_temp: 0
max_temp: 110

[fan]
pin: ar9

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AL05IRGY-if00-port0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 1500
max_z_velocity: 10
max_z_accel: 50
square_corner_velocity: 5

[safe_z_home]
home_xy_position: 150,150 # Change coordinates to the center of your print bed
speed: 50
z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[force_move]
enable_force_move: True

[bltouch]
sensor_pin: ^ar18
control_pin: ar11
set_output_mode: 5V
x_offset: -33.0
y_offset: -2.0
speed: 10

[bed_mesh]
speed: 200
mesh_min: 50,50
mesh_max: 250,250
probe_count: 4,4

[filament_switch_sensor my_filemant_sensor]
pause_on_runout: True
pause_delay: 0.5
switch_pin: ar2

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 2.140
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  0.027500, -0.010000, 0.027500, 0.177500
#*# 	  0.050000, -0.015000, 0.022500, 0.140000
#*# 	  -0.032500, -0.057500, -0.017500, 0.080000
#*# 	  0.015000, -0.010000, 0.045000, 0.180000
#*# x_count = 4
#*# y_count = 4
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 50.0000000001
#*# max_x = 249.975
#*# min_y = 50.0000000001
#*# max_y = 249.975
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 26.636
#*# pid_ki = 1.296
#*# pid_kd = 136.840
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 67.769
#*# pid_ki = 0.881
#*# pid_kd = 1303.702
`

chunter1, 2020-06-22:
> Bed Mesh leveling works even better than with marlin

I noticed that aswell. :)

Another thing i like is that there is no crypic EEPROM storage used.
Instead, the values are appended to the config-file thus you can always have a look at the values stored.

Hoffi-D3, 2020-06-22:
> Here is my Klipper config for the CR10S Pro so far.
> 
> Please note, that I modded the TMC2209 on the Creality motherboard to support UART as described here: https://www.3dprintingforum.org/creality3d/126203-tmc2208-uart-on-cr10s-pro-for-linear-advance-guide#post126203
> 
> That way, i can use pressure advance (linear advance) on my CR10S Pro. :-)
> 
> Further, for bed leveling, i am using a BLTouch.
> 
> I highly recommend looking at the example config files Klipper provides in it's documentation.
> 
> 
> 
> 
> 
> [stepper_x]
> 
> step_pin: ar54
> 
> dir_pin: ar55
> 
> enable_pin: !ar38
> 
> step_distance: .0125
> 
> endstop_pin: ^ar3
> 
> position_endstop: 0
> 
> position_max: 310
> 
> homing_speed: 80
> 
> 
> 
> [stepper_y]
> 
> step_pin: ar60
> 
> dir_pin: !ar61
> 
> enable_pin: !ar56
> 
> step_distance: .0125
> 
> endstop_pin: ^ar14
> 
> position_endstop: 0
> 
> position_max: 310
> 
> homing_speed: 80
> 
> 
> 
> [stepper_z]
> 
> step_pin: ar46
> 
> dir_pin: !ar48
> 
> enable_pin: !ar62
> 
> step_distance: .0025
> 
> endstop_pin: probe: z_virtual_endstop
> 
> position_min: -5.0
> 
> position_max: 400
> 
> 
> 
> [extruder]
> 
> step_pin: ar26
> 
> dir_pin: ar28
> 
> enable_pin: !ar24
> 
> step_distance: .00717641
> 
> nozzle_diameter: 0.400
> 
> filament_diameter: 1.750
> 
> heater_pin: ar10
> 
> sensor_type: EPCOS 100K B57560G104F
> 
> sensor_pin: analog13
> 
> min_temp: 0
> 
> max_temp: 250
> 
> min_extrude_temp: 170
> 
> max_extrude_only_distance: 600
> 
> pressure_advance: 0.5
> 
> max_extrude_only_velocity: 200
> 
> max_extrude_only_accel: 3000
> 
> 
> 
> [tmc2208 extruder]
> 
> uart_pin: ar33
> 
> microsteps: 16
> 
> interpolate: True
> 
> run_current: 0.730
> 
> sense_resistor: 0.150
> 
> 
> 
> [heater_bed]
> 
> heater_pin: ar8
> 
> sensor_type: ATC Semitec 104GT-2
> 
> sensor_pin: analog14
> 
> min_temp: 0
> 
> max_temp: 110
> 
> 
> 
> [fan]
> 
> pin: ar9
> 
> 
> 
> [mcu]
> 
> serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AL05IRGY-if00-port0
> 
> pin_map: arduino
> 
> 
> 
> [printer]
> 
> kinematics: cartesian
> 
> max_velocity: 300
> 
> max_accel: 1500
> 
> max_z_velocity: 10
> 
> max_z_accel: 50
> 
> square_corner_velocity: 5
> 
> 
> 
> [safe_z_home]
> 
> home_xy_position: 150,150 # Change coordinates to the center of your print bed
> 
> speed: 50
> 
> z_hop: 10                 # Move up 10mm
> 
> z_hop_speed: 5
> 
> 
> 
> [force_move]
> 
> enable_force_move: True
> 
> 
> 
> [bltouch]
> 
> sensor_pin: ^ar18
> 
> control_pin: ar11
> 
> set_output_mode: 5V
> 
> x_offset: -33.0
> 
> y_offset: -2.0
> 
> speed: 10
> 
> 
> 
> [bed_mesh]
> 
> speed: 200
> 
> mesh_min: 50,50
> 
> mesh_max: 250,250
> 
> probe_count: 4,4
> 
> 
> 
> [filament_switch_sensor my_filemant_sensor]
> 
> pause_on_runout: True
> 
> pause_delay: 0.5
> 
> switch_pin: ar2
> 
> 
> 
> #*# <---------------------- SAVE_CONFIG ---------------------->
> 
> #*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
> 
> #*#
> 
> #*# [bltouch]
> 
> #*# z_offset = 2.140
> 
> #*#
> 
> #*# [bed_mesh default]
> 
> #*# version = 1
> 
> #*# points =
> 
> #*# 	  0.027500, -0.010000, 0.027500, 0.177500
> 
> #*# 	  0.050000, -0.015000, 0.022500, 0.140000
> 
> #*# 	  -0.032500, -0.057500, -0.017500, 0.080000
> 
> #*# 	  0.015000, -0.010000, 0.045000, 0.180000
> 
> #*# x_count = 4
> 
> #*# y_count = 4
> 
> #*# mesh_x_pps = 2
> 
> #*# mesh_y_pps = 2
> 
> #*# algo = lagrange
> 
> #*# tension = 0.2
> 
> #*# min_x = 50.0000000001
> 
> #*# max_x = 249.975
> 
> #*# min_y = 50.0000000001
> 
> #*# max_y = 249.975
> 
> #*#
> 
> #*# [extruder]
> 
> #*# control = pid
> 
> #*# pid_kp = 26.636
> 
> #*# pid_ki = 1.296
> 
> #*# pid_kd = 136.840
> 
> #*#
> 
> #*# [heater_bed]
> 
> #*# control = pid
> 
> #*# pid_kp = 67.769
> 
> #*# pid_ki = 0.881
> 
> #*# pid_kd = 1303.702
> 
> 

Thank you! I will try :)

daredevil-ferreira, 2020-06-23:
I added this macro to the printer.cfg file to be able to tune the Z offset.

==================================
[gcode_macro G29]
gcode:
        G1 Z10 F600
#Sheet 1#       SET_GCODE_OFFSET Z=-0.645
#Sheet 2#        SET_GCODE_OFFSET Z=-0.48
        SET_GCODE_OFFSET Z=-0.48
        BED_MESH_CALIBRATE

mihaiserban, 2020-06-23:
> I added this macro to the printer.cfg file to be able to tune the Z offset.
> 
> ==================================
> [gcode_macro G29]
> gcode:
> G1 Z10 F600
> #Sheet 1# SET_GCODE_OFFSET Z=-0.645
> #Sheet 2# SET_GCODE_OFFSET Z=-0.48
> SET_GCODE_OFFSET Z=-0.48
> BED_MESH_CALIBRATE

Use the following to calibrate z offset in the terminal.

PROBE_CALIBRATE
TESTZ Z=-.1
ACCEPT
SAVE_CONFIG

chunter1, 2020-06-23:
We should change to another location/forum concerning running Klipper on a CR10s Pro since this is not related to Insanity automation...

chunter1, 2020-06-24:
By the way..
If someone wants to replace the stock Creality LCD with a 4,3" Waveshare display to run e.g. octodash, here are the proper mounting parts;
https://www.thingiverse.com/thing:4427677

I now got Klipper running on my CR10S Pro in combination with a Raspberry Pi 3 and a Waveshare 4,3" IPS capacitive touch display.
Finally no more dependency on Creality, Marlin, etc. and no performance limits (see graph before ;-) ).

[image]

Hoffi-D3, 2020-06-24:
> By the way..
> 
> If someone wants to replace the stock Creality LCD with a 4,3" Waveshare display to run e.g. octodash, here are the proper mounting parts;
> 
> https://www.thingiverse.com/thing:4427677
> 
> 
> 
> I now got Klipper running on my CR10S Pro in combination with a Raspberry Pi 3 and a Waveshare 4,3" IPS capacitive touch display.
> 
> Finally no more dependency on Creality, Marlin, etc. and no performance limits (see graph before ;-) ).
> 
> 
> 
> [EDIT] Display is perfectly clean in reality, not as shown on the image below.
> 
> 
> 
> [image]
> 
> 

I have a few questions on your setup.

Would you help me?

Any suggestions where we could chat? Do you have a 3dprintingforum.org account?

chunter1, 2020-06-24:
>  Do you have a 3dprintingforum.org account?

Jup, what's your user name there?

yufanyufan, 2020-07-19:
I compiled a version from Marlin truck for 8bit stock board.
The display does not work, but the blob issue is gone.
So there is something wrong with this version.

DodgeDeBoulet, 2020-07-19:
The CrealityDWin2.0_Bleeding branch has been synced with bugfix-2.0.x very recently. Which branch did you use?

yufanyufan, 2020-07-19:
I used CrealityDWin_2.0 and had the blob issue. I didn’t try CrealityDWin2.0_Bleeding.

InsanityAutomation, 2020-07-19:
The issue is caused by Junction deviation, which was made the default awhile ago in Marlin 2.0 Bleeding has disabled this and gone to Classic Jerk which will be in the next release tagged.

closed 2020-07-19 by InsanityAutomation