InsanityAutomation/Marlin #136: Print does not start on CR-10S Pro

Aron84, 2020-05-10:
### Description

### Steps to Reproduce

1. I try to Start a print

It should heat up, probe the bed and start the print.

Instead it shows cooling and does nothing.

#### Additional Information

It has printed before but it looks like it is not reliable.
I tryed DW6 and DW6.2 but i think i stick to original Creality firmware till this is fixed.

jesusbackflips, 2020-05-11:
Maybe you enabled Filament Runout Sensor without have one?
Or missed you need for CR10S flash your Display?

thetailwind, 2020-05-21:
I believe I am exhibiting the same symptom. @Aron84 can you elaborate why you think it's a cura problem? 

Using the exact same Gcode sliced from cura 4.5, on the Tinymachines firmware, It would start the print, do the filament purge and swipe and about 75% though the swipe process, it would just stop the steppers and display a black, but backlit screen. It was unresponsive other than power-cycling it. Octopi log reported a failure to level the printer. It was too generic to get anything useful.

On the latest Creality firmware it works on the same gcode as the tinymachines firmware. 

What did you do to cura to fix this issue? 

I

Aron84, 2020-05-21:
I did flash it back to original firmware and then to version 6.2 and now it works. I have a raspberry pi 3b conected now with no problems but i dont know why it has not printed first but I tested it with a pre sliced file from Creality when the printer won´t print and the pre sliced file worked. I thought this must be a problem with cura but i´m pretty new to 3d printing so i don´t know what was the problem.

boggisthecat, 2020-05-29:
I had this stuck ‘Cooling’ message problem as well.  After lot’s of messing around, it turned out to be an issue with long filenames corrupting the display software somehow.  (Other faults could also occur, but the stuck ‘Cooling’ was the most common.  All problems were repeatable using the same file.)

The solution was to shorten filenames.

Looks like some sort of buffer overrun type of problem.

elecktron3d, 2020-06-03:
@Aron84 something similar happens on my printer, i can print using PLA heating under 210 °C but i can not print PETG cause the printer CR10S-Pro can not goal 235 °C, the thermal runaway shows up.
What do you recommend? flash the official firmware and then the DW6.2 TM?
Did you change something on cura?

dhbiker, 2020-07-27:
The printing issue is when you have too long name of the gcode file you are trying to print.... i noticed this with the original firmware first and this firmware is also acting the same.

It has nothing to do with the slicer setttings just the pure gcode filename ;)

InsanityAutomation, 2021-04-28:
file name handling has been rewritten, so im closing this for now as related to the string handling there. If any of the limitations from here still persist, please open a new issue.

closed 2021-04-28 by InsanityAutomation