InsanityAutomation/Marlin #138: Not an issue, more of a request

cuboy29, 2020-05-14:
Using the SKR1.4 Turbo with the stock Ender 5 Plus screen AND a microSD extension to bring the microSD to the outside of the case (didn't want to modify the case).   The system does not know when an SD card is inserted because of the extension cable.  If I cycle the power to the printer with the SD inserted then everything is fine (it can see the contents of the card).  If I send an M21 command from the terminal, the card refresh and the contents show up on the LCD.  

Request:
Can the screen FW be modified to send "M21" to refresh the SD when the user pushed the "Print" button from the main screen?

Is that doable?

Thanks

cuboy29, 2020-05-14:
I guess this might force an update.

        RTS_SDCardUpate(false, true);

InsanityAutomation, 2020-07-19:
I believe at one point it refreshed at screen entry as many had damaged sd detect pins anyway. Ill take a look at some point.

InsanityAutomation, 2021-04-29:
Closing as its included in DW7.3.1 which is building now to post today.

closed 2021-04-29 by InsanityAutomation