InsanityAutomation/Marlin #140: Platformio - "MachineCR10SPro" always "defined"?

chunter1, 2020-05-15:
When I look at the latest "CrealityDwin2.0_Bleeding" code for my CR10S Pro, the section below in the Configuration.h file is always shown as enabled/"highlighted" in Platformio.
I only get this section disabled if i add a "#undef MachineCR10SPro" right before.
The question is, where is the "MachineCR10SPro" defined - I did check that there is no printer defined in the Configuration.h at the top or anywhere.
Additionally, when i look at the Configuration_adv.h, there is no plausible highlighting of preprocessor directives at all?!
Is that a Platformio setting/issue or what else can i check?

> #if ENABLED(MachineCR10SPro)
>   #define MachineCR10Std
>   #if DISABLED(ABL_BLTOUCH, ABL_EZABL, ABL_TOUCH_MI)
>     #define ABL_NCSW
>   #endif
>   #if DISABLED(ABL_UBL)
>     #define ABL_BI
>   #endif
>   #define lerdgeFilSensor
>   #if DISABLED(BedAC)
>     #define BedDC
>   #endif
> #endif

DodgeDeBoulet, 2020-06-01:
I think it's because it's a the first in the list of default_envs in platformio.ini, and that there's no env_default assignment.

I created my own custom project task:

[env:E5P_UBL_DD_SKR14]
platform = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip
board = nxp_lpc1769
build_flags = -DU8G_HAL_LINKS -IMarlin/src/HAL/LPC1768/include -IMarlin/src/HAL/LPC1768/u8g ${common.build_flags} -DMachineEnder5Plus -DHotendAllMetal -DDirectDrive -DBedDC -DMelziHostOnly -DSKR_2209 -DSKR_UART -DABL_UBL -DCLASSIC_JERK -DForce10SProDisplay -DSKR14Turbo
lib_ldf_mode = off
lib_compat_mode = strict
extra_scripts = Marlin/src/HAL/LPC1768/upload_extra_script.py
src_filter = ${common.default_src_filter} +<src/HAL/LPC1768>
lib_deps = 
	Servo
	LiquidCrystal
	U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
	TMCStepper@>=0.6.1,<1.0.0
	Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/release.zip
	SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip

 and added an env_default entry just above default_envs:

env_default = E5P_UBL_DD_SKR14

Now the header files show with the appropriate active defs (based on the -DXxxx entries for its build_flags).

InsanityAutomation, 2020-07-19:
Yup, just added mega2560 as the top task so itll show highlighting for that instead which should use config files.

closed 2020-07-19 by InsanityAutomation