InsanityAutomation/Marlin #141: BLTouch Not deploying at print start _ Creality Dwin2.0

antoinelb1, 2020-05-16:
Hi, I have a Ceality Ender 5 Plus and am trying to have it run with an SKR PRO V1.1 board. 
I have successfully built and loaded CrealityDwin_2.0 and con controll all my axis.
I have the BLtouch that came with the Creality printer.  (v3.1)
I have re-loaded and re-built with the most recent file last week. 

When connected to the board, if i send a G28 or G29 command through  Pronterface or Octopi, the printer operates properly and does level/mesh(4x4) without any issue ! The BLTouch does probe whenever needed and seem repeatable from the result i got.

I have succesfully leveling my bed using Octopi/ Pronterface
I have adjusted my Z and using the touchscreen z up/down like i had to do on the default creality board and should be ready to print.  

My "main" issue is that when i send a print request ( through cura or octopi ), It heats the bed and nozzle properly, home x and y, then move to the center of the bed  and starts moving the bed up without having the BLtouch deployed!!! The nozzle then crashes my nozzle into the bed... D: 

I have also some issue with the leveling button on the touchscreen that might be related to that one... 
-Twice the leveling button did not deploy the BLTouch ( out or over 100 try ).
-The meshing button seems to send something else than G29 as it start heating my bed and nozzle, then home x,y and crashes into the bed, like it does at a print start...

I'm having a hard time figuring out whats wrong.
Has anyone had this issue before ?
Am i configuring something wrong ?
Can anybody help ? i can provide any detail you may need to help me !

masc2279, 2020-05-26:
First run M503 and see if your auto level is truly set. G29 is doing the auto level as you stated but at no point have you said after the G29 you saved the mesh using an M500.

masc2279, 2020-06-03:
I dont see any auto leveling on your console at all  here is the section showing from my setup.

Run your auto level and then right after send an M500 to save it then re run the M503 you should see an auto level section and the data from the auto level once properly configured and saved.

[image]

antoinelb1, 2020-06-07:
Hi there, thanks for the feedback.

I had never ran M500 before but the data from the meshing seemed to be stored somewhere as it is reflected in the Creality Default touchscreen. (have to home with the front screen before running G29 to be on the proper display)

I have now ran M28, M29 followed by M500.

The results when i run M503 are the following : 

Send: M503
Recv: wait
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E95.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X750.00 Y750.00 Z10.00 E75.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000.00 Y2000.00 Z100.00 E75.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P750.00 R1000.00 T300.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.06
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:  G29 W I0 J0 Z-0.11525
Recv: echo:  G29 W I1 J0 Z-0.07275
Recv: echo:  G29 W I2 J0 Z0.00575
Recv: echo:  G29 W I3 J0 Z0.09325
Recv: echo:  G29 W I4 J0 Z0.23025
Recv: echo:  G29 W I0 J1 Z-0.13175
Recv: echo:  G29 W I1 J1 Z-0.07425
Recv: echo:  G29 W I2 J1 Z0.00000
Recv: echo:  G29 W I3 J1 Z0.10575
Recv: echo:  G29 W I4 J1 Z0.23850
Recv: echo:  G29 W I0 J2 Z-0.14250
Recv: echo:  G29 W I1 J2 Z-0.09650
Recv: echo:  G29 W I2 J2 Z-0.00525
Recv: echo:  G29 W I3 J2 Z0.08700
Recv: echo:  G29 W I4 J2 Z0.24350
Recv: echo:  G29 W I0 J3 Z-0.21525
Recv: echo:  G29 W I1 J3 Z-0.18675
Recv: echo:  G29 W I2 J3 Z-0.11350
Recv: echo:  G29 W I3 J3 Z-0.01775
Recv: echo:  G29 W I4 J3 Z0.16050
Recv: echo:  G29 W I0 J4 Z-0.28575
Recv: echo:  G29 W I1 J4 Z-0.27450
Recv: echo:  G29 W I2 J4 Z-0.22550
Recv: echo:  G29 W I3 J4 Z-0.12950
Recv: echo:  G29 W I4 J4 Z0.05675
Recv: echo:; PID settings:
Recv: echo:  M301 P14.72 I0.89 D61.22
Recv: echo:  M304 P690.34 I111.47 D1068.83
Recv: echo:; Retract: S<length> F<units/m> Z<lift>
Recv: echo:  M207 S3.00 W13.00 F2700.00 Z0.00
Recv: echo:; Recover: S<length> F<units/m>
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:; Auto-Retract: S=0 to disable, 1 to interpret E-only moves as retract/recover
Recv: echo:  M209 S0
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-41.00 Y-8.00 Z0.00
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X730 Y730 Z850
Recv: echo:  M906 T0 E730
Recv: 
Recv: echo:; Hybrid Threshold:
Recv: echo:  M913 X152 Y152 Z10
Recv: echo:  M913 T0 E50
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L600.00 U75.00
Recv: echo:; Backlash compensation:
Recv: echo:  M425 F0.00 X0.00 Y0.00 Z0.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 D10.00
Recv: ok P15 B7
Recv:  T:26.22 /0.00 B:25.31 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: wait

I did try to print but faced the exact same problem as before. 
When homing at the start of the print, the x and y homed properly, the head moved to the BLTouch to the center, DIDN't deploy the BLTouch and raised the bed until it would have crashed if not stopped ( happened before)

antoinelb1, 2020-06-15:
HI guys, 

I did a littl interresting test. 
When I start a print without any G28 command in the start Gcode, the printer works !!

Of course i had manually pushed a G28, G29 and M500 command before starting the print.

Interesting thing that happened before the successful print;
-Once i have had the printer reporting a probing fault at the first leveling of a G29 sequence. The printer then homed the X and Y went fo home Z but did not release the BL Touch... would have crashed. 
-After stopping a first succesfulle print start ( to finsetune z offset ) i have been surprise to see that the printer can no longer properly G28... did the same as it was doing when doing a G28 at print start.

masc2279, 2020-06-15:
Did you customize the firmware ?  make sure you have

antoinelb1, 2020-06-15:
Yes, i confirm that it works if i do not have the G28 command in my start Gcode, it is functional. 

I believe i only modified the minimum parameters to get it operational. 
I thought it could be something i change by mistake so i did build it 3 separate time, from different download, at different time, and ended-up with the same problem. 
Last download is about a month old...

I do have #define RESTORE_LEVELING_AFTER_G28 enabled in my current configuration.h

InsanityAutomation, 2020-07-19:
It sounds like a probe fault that isnt being detected by firmware. Im not sure theres much that can be done with it if the probe is faulting but not reporting the faulted state. Is the red light on it solid when or flashing when it fails to deploy?

antoinelb1, 2020-09-04:
No red light flashing.
As mentionned it really is like its going ot probe the bed but does not read the input...

It never happens when running the mesh through Octoprint.

InsanityAutomation, 2021-04-28:
Im going to close this old one, The issue is in areas outside where this project diverges from head, and utilizes a board that I dont have. There is a good chance its been resolved upstream and was included in subsequent builds.

closed 2021-04-28 by InsanityAutomation