InsanityAutomation/Marlin #144: Trying to use for CreatBot

ModMike, 2020-06-10:
I would like to use this branch with a Creatbot. It uses a DWIN DMT48270M043_02WT.

I created a pin file and everything works except the screen. What screen firmware would you recommend for dual head machine with no auto level? I do need load and unload for both extruders.

Thanks!

InsanityAutomation, 2020-06-11:
The CRX stock screens should be close. May need to check OS version on the panel itself to see if needs modified screen change commands like advi uses. They run the prior generation dwin firmware on the screen itself.

dragonpaco, 2020-12-10:
> 
> 
> I would like to use this branch with a Creatbot. It uses a DWIN DMT48270M043_02WT.
> 
> I created a pin file and everything works except the screen. What screen firmware would you recommend for dual head machine with no auto level? I do need load and unload for both extruders.
> 
> Thanks!

Did you ever get Marlin working correctly on the creatbot dx? Creatbot is refusing to share their code and the lcd/usb issue is not worth the benefits of BLtouch.  

Thanks

ModMike, 2020-12-11:
I did get ti working somewhat using a Frankenstein version. The screen partially works but code needs to be gone through to make all the commands work.

I don't even remember what I did.

InsanityAutomation, 2020-12-11:
Get someone to send me one and ill make it run lol

ModMike, 2020-12-11:
Whats your shipping address?

InsanityAutomation, 2020-12-11:
Shoot me an email - d.menzel@insanityautomation.com

InsanityAutomation, 2021-06-21:
FYI, current screen builds add manual mesh leveling for DW74 and may just work as is...

InsanityAutomation, 2023-02-17:
Never heard more on Creatbot or got one. Closing

closed 2023-02-17 by InsanityAutomation