InsanityAutomation/Marlin #147: Octoprint unable to control CR10sPRO during M600

pricedev, 2020-06-25:
### Description

When executing the M600 command and using Octoprint, the printer shows messages to the Octoprint web page, but those messages are not actionable, and the only way I've found to execute a filament change is solely using the printer LCD panel.

After discussing the situation on the Octoprint Discord server, the recommendation was to restore the pause / resume / cancel defines to the configuration_adv.h file:

#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
  #define ACTION_ON_PAUSE "pause"
  #define ACTION_ON_RESUME "resume"
  #define ACTION_ON_CANCEL "cancel"
#endif

From reviewing the TM 2.0 configuration_adv.h file for Marlin-CrealityDwin_2.0-2019-11-04, I see the following:

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #if DISABLED(MachineCR10Orig)
    #define HOST_PROMPT_SUPPORT
  #endif
#endif

The recommendation is to restore the pause / resume / cancel defines, since they are also required for host control.  Indications are that the main Marlin firmware repo had those lines mistakenly removed.

### Steps to Reproduce

Steps to change from the LCD Panel:
1.  M600 command executed
2. Nozzle parks and printer unloads approximately 80 mm of filament, that gets it out of the hotend but leaves it fully extended through the PTFE Bowden tube
3. LCD displays "no filament - Yes to change"
4. Printer hotend very quickly starts to cool down (no indication of that on the LCD due to the no filament dialog, and no indication in Octoprint due to the dialogs displayed there, aside from "reheat timeout")
5. At this point if I hit YES on the LCD panel, the hotend starts to reheat, and then very quickly starts to cool down again
6. Release the extruder arm and manually pull out the filament.
7. Load new filament - here is where hope and faith enter.  If I manually load the filament back to where I unloaded it from (ie. to the top of the heat break), when I hit "YES" on the LCD dialog, after reheating, (if a reheat is required, I have to hit YES again), the printer evidently attempts to rapidly load through the PTFE tube, which results in nasty grinding/skipping.  If I load the filament just to the start of the PTFE tube, the majority of the grinding / skipping is avoided, but I stand a good chance of not getting a pure new color.  
8. As soon as purging is completed, the print resumes.

While all this is going on, there are various messages about nozzle parked / paused / reheat timedout / and a curious "Purge more" dialog.  The problem is that the only way I've found to initiate the reheat is from the LCD panel.  Once the nozzle is back up to temp, I'm not sure the Octoprint dialogs will do anything, and if you don't act very quickly the nozzle starts cooling again.  So, the only way to progress is to hit "YES" again on the filament change prompt on the LCD Panel, and at that point you are locked in to the filament load, followed immediately by resuming printing.

So, at this point, I think the M600 only works from the LCD Panel (as far as I've been able to determine).  While it technically does change the filament, getting a pure color change is a somewhat random event.

**Expected behavior:** [What you expect to happen]

1. Printer stops printing and moves head to rest position
2. Printer unloads filament and waits
3. Operator removes filament and uses gcode originating from the host (Octoprint) to reheat hotend (if needed), load filament, extrude new filament (using host originated gcode scipts) to verify a clean color change
4. Operator signals printer to return to printing (either through host commands or by pressing the resume button on the printer LCD panel.

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thsrp-bob-roger, 2020-07-03:
For your information, the settings are unload 75mm and reload 430mm. !! ? I do not know why? You can see these settings in the eproom and therefore you can change them

pricedev, 2020-07-03:
Thanks,  I'll give that a shot.  I might give that a shot, but still have
the issue that it just loads and goes without confirming the color change
is clean.  If I have time, I may also try to compile in the additional
define's that I mentioned.

pricedev, 2021-04-28:
> The issue here is actually notably deeper. There is currently no state
> structure for determining the actual position in the process that is
> accessible to touchscreens. A fairly large undertaking is needed to resolve
> this correctly.

Thanks for checking in.

I've actually found another solution that works exceptionally well, so
support for M600 is no longer a necessary piece of functionality, for me at
least.

My alternative solution is a collection of pause and resume macros
available for Octoprint.  Octoprint's hooks for scripts on pause and
resume, coupled with starter scripts they provide in their wiki, solve the
situation perfectly.  It is all driven through Octoprint, but that fits my
workflows better anyway.

Thanks for all your work on this firmware.

Ray

InsanityAutomation, 2021-04-28:
Ok, im going to close this and keep #146 as the only M600/M125 issue open.

closed 2021-04-28 by InsanityAutomation