InsanityAutomation/Marlin #149: Cr10s pro shut down recovery still dosent work

Nicolozano97, 2020-07-20:
### Description

I´ve been trying the lastest firmware for the cr10s pro with bil and the shut down recovery dosn´t work, I always lost the print with the new version. i will be gratefull if you can solve it.  I have the lastest screen firmware too.  

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mrbipbip, 2020-08-14:
Also the same issue here. Since Marlin 2.0 (with the stable branch too).

InsanityAutomation, 2021-04-28:
Please keep in mind it is included but disabled by default due to reported crashes with many SD cards. If you want to have PLR active for youre print, youll need M413S1 in youre starting gcode. If the printer reboots immediately after starting a print, use a different SD card.

closed 2021-04-28 by InsanityAutomation