InsanityAutomation/Marlin #150: Reqest : CR10s Pro V2 Hexfile with Slice engineering Heater + thermistor and updated homing and Finish print homing

kot0005, 2020-07-30:
Request for a modified Hexfile ( http://insanityautomation.com/Firmware/Creality/10SProV2_BIL_DW6.2.hex.gz ) so it uses Slice engineering's Thermistor and Heater cartridge (24v and 50W) with a Mosquito hotend.

I would also like to request for fixes from this issue to be included in it please:  https://github.com/InsanityAutomation/Marlin/issues/120

I have a CR10s Pro v2 and am currently using the hexfile linked above as my firmware, however I would like to use Slice engineering's hotend with their heater and thermistor.

Thank you!

kay0815, 2020-08-04:
I also would like to combine my CR10sProV2 with the Mosquito hotend, thermistor and heating cartridge from SLICE engineering and would appreciate a hex file for this combination.
Maybe someone can help me with the necessary settings and compiling a marlin 2.0 based firmware for this special printer? Unfortunately within the examples there is no CR10sproV2 and I don't know which changes have to be made when using a CR10 or CR10S as a base.
Thanks a lot.

best regards, Kay

kot0005, 2020-08-04:
> I also would like to combine my CR10sProV2 with the Mosquito hotend, thermistor and heating cartridge from SLICE engineering and would appreciate a hex file for this combination.
> Maybe someone can [...]

same here.

kot0005, 2020-08-27:
> I also would like to combine my CR10sProV2 with the Mosquito hotend, thermistor and heating cartridge from SLICE engineering and would appreciate a hex file for this combination.
> Maybe someone can [...]

hey, someone modified and compiled a hexx file for cr10s prov2 to include the slice engineering tables

https://www.dropbox.com/transfer/AAAAAOMcPbsWxdfyCQ3iCxQBIdbdU6mw8gm9IVWo6oRj_d4km87mLVw

are you able to test it ? i dont have my thermistor and heater yet.

I also asked them for the marlin file 

https://www.dropbox.com/transfer/AAAAAO6E-G4wJK22MmHEd_480CXkA6uhP-cMlRVMgIIGYz_UCNTFx-Q

kay0815, 2020-09-18:
sorry, tried the dropbox links to late, they do not work anymore. Did you have success getting the marlin files yet? I did not want to order parts without beeing sure having a firmware to get them working, so I also do not have the mosquito yet. Sorry.
best regards, Kay

kot0005, 2020-09-24:
I got a Hex file for CR10s pro v2 with Mosquito, slice heater and Thermistor from Tinymachines, heaps thanks to them !! It doesnt have the homing fix but oh wells.

kay0815, 2020-10-01:
well, is it included within the hex-file section?

kot0005, 2020-10-11:
> well, is it included within the hex-file section?

no.. i got it from emailign support however the firmware has issues, like pause function doesnt work, printer wont resume if you pause. Same with filament change. Printer wont resume printing if filament runsout..

InsanityAutomation, 2021-04-28:
These have files have been included in the last several iterations
MC = Mosquito Creality mount
ME = Mosquito E3D mount

closed 2021-04-28 by InsanityAutomation