InsanityAutomation/Marlin #154: CR-X (CRX_BLT_UBL_Fil_DW7) problem with serial connection and OctoPrint

LMS0815, 2020-09-27:
### Description

After updating my Creality CR-X from **CRX_BLT_UBL_Fil_DW6.2** to **CRX_BLT_UBL_Fil_DW7** OctoPrint is having issues connecting to the printer. Printing is impossible.

### Steps to Reproduce

1. Use CRX_BLT_UBL_Fil_DW7.hex
2. Try printing with octoprint (PI B+ Installation)

#### Additional Information

reverting to **CRX_BLT_UBL_Fil_DW6.2** solves the problem.

pfarmer1979, 2020-09-30:
I upgraded my printer to 10SPro_BIL_DW7 from an old 6.x version today and I found Octoprint didn't connect using the previous working baud rate of 115200. Using 250000 did the trick for me. It's doing it's first print right now and seems OK so far.

LMS0815, 2020-09-30:
Thank you. I tried 250000 as well (Which was my default anyway) but no success.

LMS0815, 2020-09-30:
Just to be more precise. I can connect to the printer but I get a timeout if I am printing:

```
Send: N2 M420 S*84
Recv: echo:No SD card
Recv: //action:notification SD Init Fail
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N3 M105*36
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N4 M105*35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N5 M105*34
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N6 M105*33
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N7 M105*32
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Starting" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

```

pfarmer1979, 2020-10-09:
Since my last post - I'm now experiencing the same issue with no change to Octoprint or my CR10s. My logs look pretty much the same as yours LMS0815. I'll revert back to 6.2 later today I think.

LMS0815, 2020-10-09:
Seems printing from SD is as well not working with Version 7.0 on CR-X

alex-murphy, 2020-10-10:
I'm having the same issues with 6.2 firmware as well. I can print files locally from octoprint but not from the SD-card. When printing from SD the printer starts the heating function then locks up which was the same issue as I experience in v7, then octoprint would timeout. Anyhoo, at least I can print :)

InsanityAutomation, 2021-04-27:
Can you confirm if the rewritten file browser resolves the naming issues described as well as any changes to serial stability on 7.3? I have not been able to reproduce this kind of communication dropout here.

stooged, 2022-10-22:
> I upgraded my printer to 10SPro_BIL_DW7 from an old 6.x version today and I found Octoprint didn't connect using the previous working baud rate of 115200. Using 250000 did the trick for me. It's doing it's first print right now and seems OK so far.

you resolved the same issue i was having with the current firmware, the baud rate was the trick, thankyou.