InsanityAutomation/Marlin #155: BLTouch flashing red after flashing to new Marlin firmware (DW7)

zygote279, 2020-09-30:
Hi all - I flashed my CR-10 Max to the DW7 firmware and all completed fine except my BL Touch no longer works - it just flashes red on boot. Seems to be permanently set to 'triggered' and can't reset it through GCODE etc.

Is there anything obvious I should check here - I've ordered a new BLTouch just in case, but struggling to see the coincidence of precisely the moment the software is upgraded, the hardware fails! (I wasn't touching/changing anything hardware related at the time, simply new firmware)

Appreciate any thoughts or daft things I might have missed!

zerox1212, 2020-12-09:
@zygote279 I have the same problem with my Max when the machine powers on. However it doesn't always happen. At this time I don't know what triggers the issue.

zygote279, 2020-12-09:
> @zygote279 I have the same problem with my Max when the machine powers on. However it doesn't always happen. At this time I don't know what triggers the issue.

For me, replacing the BL touch did the trick, obviously a dodgy one....! Pretty quick to swap out

zerox1212, 2020-12-09:
I would double check that, I think the bug with BLTouch flash happens only after an EPROM reset, and before you level the bed. At least in my case the flashing BLTouch recovered after going to the level screen and proceeding with bed leveling process.

zygote279, 2020-12-10:
Interesting - pretty sure it was faulty (literally swapped the component and immediately has worked fine ever since), but might have a mosey if I get a spare moment over xmas to see if this could be related (tried resetting bltouch in gcode but this did nothing)

InsanityAutomation, 2021-04-28:
A quick note to add, Clones can fault with the commands to set modes on BLTouch V3 probes, wheras older genuine ignore them nicely. As a replacement resolved it, im closing this out.

closed 2021-04-28 by InsanityAutomation