InsanityAutomation/Marlin #156: E5P Pause error (no filament)

Frazor13, 2020-10-04:
### Description

 I set up E5P_BIL_DW7.hex to my stock Ender 5 Plus. 

Everything works fine, but when I press the pause button on printer screen it says no filament as you can see in the picture. When I touch YES nothing happens and it does only react to NO, resulting in an aborted print.. Same happens when I use the pause at height function in cura. It pauses, but doesn't resume.

At first I thought it could be caused by the filament runout sensor, but it works fine and all cables are connected properly. I also tried to disable the sensor with M412 S0 and removed the cable but I still get this error.

[image]

gtenney99, 2020-10-07:
Im experiencing a similar problem when the filament runout sensor triggers, pressing yes does nothing other than turn the nozzle heat back on (verified with octoprint), while no results in a failed print.  Behavior doesn't change if i try and press yes without loading new filament, or after i load new filament.

Im using E5P_BIL_DW7.hex on stock machine.

thsrp-bob-roger, 2020-10-09:
Yes the M600 function (change filament) and the "filament runout sensor" (which shares a large part of the M600 script) do not work in the DW7. There is a bug. I return to DW6.2 while waiting   . CR10S Pro

Frazor13, 2020-10-12:
> Yes the M600 function (change filament) and the "filament runout sensor" (which shares a large part of the M600 script) do not work in the DW7. There is a bug. I return to DW6.2 while waiting . CR10S Pro

Where can I find the old DW6.2?

crackness, 2020-11-19:
Confirming this is an issue for me as well.

FrancescoTalotta, 2020-11-20:
Yes, me too I have got the same problem.

lukasbohm, 2021-01-24:
Hi, so the exchange doesn't work for me either, and the protection against switching off the power doesn't really work for me, does anyone know what to do with it? and does anyone have the old version 6.2? Thanks****

Uforias, 2021-01-24:
> 
> 
> Hi, so the exchange doesn't work for me either, and the protection against switching off the power doesn't really work for me, does anyone know what to do with it? and does anyone have the old version 6.2? Thanks****

Here it is: https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin6.2

Uforias, 2021-01-24:
> 
> 
> > Yes the M600 function (change filament) and the "filament runout sensor" (which shares a large part of the M600 script) do not work in the DW7. There is a bug. I return to DW6.2 while waiting . CR10S Pro
> 
> Where can I find the old DW6.2?

Here: https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin6.2

lukasbohm, 2021-01-24:
> > > Ano, funkce M600 (změna vlákna) a „snímač házení vlákna“ (který sdílí velkou část skriptu M600) v DW7 nefungují. Je tu chyba. Během čekání se vracím k DW6.2. CR10S Pro
> > 
> > 
> > Kde najdu starý DW6.2?
> 
> Zde: https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin6.2

Thank you as soon as I'm home from the night download and upload. I admit that it happens to me when the exchange or current monitoring doesn't work for me :-(. To put it together.

InsanityAutomation, 2021-06-20:
As this has been largely redone, and even further improved in the 7.4 update thats in final testing, im closing the older M600 issues.

closed 2021-06-20 by InsanityAutomation