InsanityAutomation/Marlin #158: issue with SKR v1.2 e3

numpty9989, 2020-10-13:
### Description

Trying to create for the SKR v1.2 e3 but I only see V2 option.. I have tried and all o get is a lit blank screen

### Steps to Reproduce

1. compile the firmware
2. insert SD
3. Turn on

**Expected behaviour:** 
working screen usable printer

**Actual behaviour:** 
blank screen with backlight

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-
[Firmware files.zip](https://github.com/InsanityAutomation/Marlin/files/5374643/Firmware.files.zip)
[image]

can-i-contribute) for additional guidelines.

quarky42, 2020-10-14:
Same issue here.  Ender 3 with SKR v1.2 E3

Latest version pulled from here:
https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

Trying to compile this configuration:
[env:STM32F103RC_btt_512K_USB_Ender3]
platform = ${common_stm32f1.platform}
extends = env:STM32F103RC_btt_512K
build_flags = ${env:STM32F103RC_btt_512K.build_flags} -DUSE_USB_COMPOSITE -DSKRMiniE3V2 -DMachineEnder3 -DHotendAllMetal -DBedDC -DPOWER_LOSS_RECOVERY -DABL_BLTOUCH
lib_deps = 
	${env:STM32F103RC_btt_512K.lib_deps}
	USBComposite for STM32F1@0.91

The above is a copy of the [env:STM32F103RC_btt_512K_USB] block with these flags added:
-DSKRMiniE3V2 -DMachineEnder3 -DHotendAllMetal -DBedDC -DPOWER_LOSS_RECOVERY -DABL_BLTOUCH

This results in a compiled firmware that gives a blank screen when the firmware is loaded and a non-functioning printer.   The printer had previously a Marlin 2.0.x version loaded from marlinfw.org but was exhibiting problems with first layer not showing evidence of the bed mesh being used.   I was not expecting to only have a SKRMiniE3V2 flag.  I thought there would be a SKRMiniE3 flag, but didn't see one.

I tried manually replacing the #define MOTHERBOARD line to point to the SKR Mini E3 1.2 board, but it errored out on that as well.

I tried manually forcing the CR10 Display type among other things.

I am using the latest PlatformIO and Visual Studio Code.  All libraries / platform packages were up to date.  I tried to roll back to an earlier version of PlatformIO but I was not able to get a good test of that.  The reason I tried this was because of a past bug in Marlin just in case this issue had made its way over to InsanityAutomation's code:
https://www.reddit.com/r/ender3/comments/dtk7yh/blank_screen_skr_mini_e3_v12/

quarky42, 2020-10-14:
It appears that using the -DSKRMiniE3V2 flag is wrong.   That flag sets the V2 board as being used, and I can see that the LCD pin assignments are different between the V1.2 and V2 boards.   So I have manually gone through and set the V1_2 board as being used.   The code immediately complains about USE_CONTROLLER_FAN, so I temporarily comment that out.   Looks like the V2 is setup to use CONTROLLER_FAN_PIN  FAN1_PIN.  Well, the V1.2 also has that pin, but it isn't in the board assignment.   

It seems to me that the code isn't setup for the V1.2 at all and it is going to take a lot of manual configuration.  I've attempted it, but we still have no display.   I'll continue looking at the info I can dig up on the pin assignments and see if there is anything else I can do.  

I have a CR10S Pro with SKR V1.4 Turbo board using the stock touch screen display using this type of setup where I combine the flags for the various elements.  It works very well.  I was really disappointed when I couldn't find a proper set of flags for the Ender 3 / SKR Mini E3 V1.2 board as that is a dedicated combination.... that is a drop in replacement board, so either I'm really missing something fundamental here, or there is a major combination that just isn't setup fully.   I am an amateur when it comes to dealing with PlatformIO, so there is also a large possibility that I'm just missing a really important flag or something there?

quarky42, 2020-10-14:
The screen works when loading the stock BTT firmware bin with bltouch from here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

quarky42, 2020-10-14:
I think the problem might be the user told me he had an Ender3, but now he's telling me) and I see at the top post, that it is in fact an Ender 3 Pro.  So, I'm basically starting all over again, but this time with the correct printer name.  So everything I've written up to this point might just need to be deleted.  I recall the Ender 3 Pro had a different screen.

I've tried just changing the machine over to -DMachineEnder3Pro422 and still had no display though.   I've got so many changes in this directory, I need to switch back to the unchanged directory and try it again there. 

So, I'm really not sure where the problem is now and I basically get to start all over again.

quarky42, 2020-10-14:
I've ran through everything again and still no display on the Ender 3 Pro with SKR Mini E3 V1.2 board.  Attached are the platformio.ini, and configuration.h  and configuration_adv.h.

The stock firmware from BTT's github has the display working on it and so does stock Marlin when it is manually configured.

In Configuration.h I changed the line that pointed to the V2.0 board over to the V1.2 board instead so that the pin assignments would be correct.   I got an error on the fan controller like expected, so I commented the use_fan_controller line out of configuration_adv.h.  

If this is a user issue and I'm doing something wrong, I apologize.  I tried setting this up the same way I've setup my SKR V1.4 Turbo by using a build flags to specify the configuration.  It works great for my CR-10S Pro w/ SKR V1.4 Turbo, so I expected it to work here with minimal change to the source code, especially since the SKR E3 Mini V1.2 board is such a popular Ender 3 / Ender 3 Pro upgrade.

[config files.zip](https://github.com/InsanityAutomation/Marlin/files/5380037/config.files.zip)

quarky42, 2020-11-02:
I've gone through this a a number of times and even attempting to configure the SKR Mini E3 V1.2 board manually through Configuration.h and _adv.h I find that the printer locks up on boot / no display.   I believe it is related to the 2.0.6 code-base and that the problem has been fixed in the latest Marlin: 
https://github.com/MarlinFirmware/Marlin/issues/16509

I just discovered this and have ran out of steam. 

I believe the following PlatformIO.ini and Config changes should work for an Ender3 using the SKR Mini E3 V1.2 Board and with an added BLTouch.  These changes do not work in DW7, but maybe these changes would work in a 2.0.7.x codebase once it gets there.

[SKRMiniE3V12_DW7_ConfigFlags.zip](https://github.com/InsanityAutomation/Marlin/files/5478599/SKRMiniE3V12_DW7_ConfigFlags.zip)

InsanityAutomation, 2021-04-29:
Keep in mind there is a reason I have never supported the SKRMini boards. I begrudgingly added the V2. To date the only one I can actually recommend is the Turbo as it has the LPC1769 MCU. One of the big reasons is the MCU on them. It is rated for 256k. The fab process for the chips is the same plate for 256k, 384k and 512k is run identically. They expect transistor failure during manufacture. If it fails tests for 512k its marked 384k, and even failing that is 256k. Anything below that threshhold is scrapped. Depending how bad the chip is and where the failures are, forcing it into rejected memory can exhibit all kinds of issues. Reports have ranged wildly from hangs with heats left on at full output to simply math errors with mesh leveling causing it to never repeat. Regardless using this at 512k is asking for problems.

Couple that with the vreg overheating (and in some cases bursting into flames) as well as lack of copper on the tmc drivers for these boards, I avoid them and support for them as much as I can.

closed 2021-04-29 by InsanityAutomation