InsanityAutomation/Marlin #159: SD card reader ....After ejecting a SD card "Media Removed" the card reader does not recognize a new card inserted into the SD card reader.

Volk53, 2020-11-04:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Volk53, 2020-11-05:
I changed the firmware to Tiny Machine "10S_Pro_BLT_BIL_DW7.hex" and it solved the problem with changing the SD card and some other related issues.

InsanityAutomation, 2021-04-28:
Closing this as the file handling has been revamped completely.

closed 2021-04-28 by InsanityAutomation