InsanityAutomation/Marlin #160: Auto leveling .. even with Auto levelling turned of the printer is always probing the 25 points before commence printing (10SPro_BLT_BIL_MC_DW7.hex)  ..any thoughts?

Volk53, 2020-11-04:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Volk53, 2020-11-04:
ok .. I solved that issue by disabling G29 in the start G-code.

closed 2020-11-04 by Volk53

LMS0815, 2020-11-05:
`G29` behaves differently depending on the ABL method

See Marlin: 
[Bed Leveling (Bilinear)](https://marlinfw.org/docs/gcode/G029-abl-bilinear.html)
[Bed Leveling (3-Point)](https://marlinfw.org/docs/gcode/G029-abl-3point.html)
[Bed Leveling (Unified)](https://marlinfw.org/docs/gcode/G029-ubl.html)