InsanityAutomation/Marlin #161: DW7 - I2C_EEPROM Not Working - SKRV14Turbo CR10SPro

quarky42, 2020-11-17:
### Description

On DW6, the I2C EEPROM support was working.  My bed leveling measurements and other configs were saving to EEPROM with no problem.   

After loading DW7, no more EEPROM functioning.

Printer:  CR10S Pro, SKR V1.4 Turbo Board, (other options can be seen below in the patformio.ini snippet I've included.

### Steps to Reproduce

1. GitHub Pull today 2020-11-16, DW7, copy the LPC1769 section in platformio.ini and add the build flags for my printer to it.
2. Flash the touch screen DWIN_SET package and flash the firmware using this custom platformio.ini build:

[env:LPC1769_CR10SPro_SKRV14Turbo]
platform  = ${common_LPC.platform}
extends   = common_LPC
board     = nxp_lpc1769
build_flags = ${common_LPC.build_flags} -DMachineCR10SPro -DHotendAllMetal -DBedAC -DABL_BLTOUCH -DPOWER_LOSS_RECOVERY -DHotendMosquito -DBondtech -DForce10SProDisplay -DSKR14Turbo -DI2C_EEPROM -DSKR_2209 -DSKR_UART

I had to create the build_flags = ${common_LPC.build_flags} line because it was missing from the LPC1769 section.

3. Boot Printer / Initialize EEPROM
4. Change any settings that should normally save to the EEPROM and press the onscreen Save Button.
5 Wait a few seconds and power cycle the printer, nothing is saved anywhere.  No bed leveling data, no steps/mm calibrations, and no probe / nozzle offsets... nothing gets saved between power cycles.

**Expected behavior:**

I expect to be able to have my BLTouch measurements and steps/mm settings changes saved between power cycles.  I used a very similar process to in DW6 to create my firmware and it works great.

**Actual behavior:**

Settings aren't saved and revert back to the default.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
Configuration.h and Configuration_adv.h are unchanged and not uploaded at this time.  My platformio.ini section is pasted above.

* Provide pictures or links to videos that clearly demonstrate the issue.
I don't see the value in taking pictures of settings that are being applied and then disappear, but if you need this, let me know and I will provide.

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

quarky42, 2020-11-17:
Is this helpful: 
https://github.com/MarlinFirmware/Marlin/issues/17799

I know it is the SKR V1.1 Pro, so there may be some differences, but just the same there may be some similarities as well.

and this one more specific to the SKR V1.4:
https://github.com/MarlinFirmware/Marlin/pull/17858

quarky42, 2020-11-17:
Adding -DSDCARD_EEPROM_EMULATION  instead of -DI2C_EEPROM gets my printer working using the SDCard, however I would much rather use the dedicated EEPROM that I have in this printer / that works with DW6.

I've tried manually turning on a number of flags I found related to the I2C EEPROM and was not able to get it to work.

ebrbt, 2020-12-10:
I was facing the same issue.
It seems to be solved by commenting out // #define PRINTCOUNTER
I also have commented out //  #define EEPROM_AUTO_INIT

Hope this helps

quarky42, 2020-12-10:
I'm really looking forward to trying this.  Thank you!

> I was facing the same issue.
> It seems to be solved by commenting out // #define PRINTCOUNTER
> I also have commented out // #define EEPROM_AUTO_INIT

quarky42, 2020-12-12:
Looking at the EEPROM_AUTO_INIT line the comment is:  "Init EEPROM automatically on any errors."

If that is causing the eeprom to be wiped every time the system is booted up, that might explain the behavior.  I'd have to go connect up a computer to the printer and right now it's pretty much standalone until I maybe setup another octoprint server for it.

I don't see why PRINTCOUNTER would be part of it, but I could certainly see how EEPROM_AUTO_INIT, something is triggering and wiping the firmware in this situation.

ebrbt, 2020-12-14:
This solution comes from [there](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/127) .
Can't explain it, but for me the PRINTCOUNTER make the EEPROM persists after power cycle.

quarky42, 2020-12-14:
Thank you for the added background.  Seems like people might be having mixed results with it maybe.  I'll try it later, for sure, then.

quarky42, 2020-12-14:
Turning off only PRINTCOUNTER, by itself does not fix the issue.   If I turn off the EEPROM_AUTO_INIT that is likely to mask / hide the issue.  Something is likely corrupting the I2C EEPROM and the firmware is detecting it and resetting the EEPROM for me.

elik745i, 2020-12-15:
What is the difference between DW6 and DW7? what advantages. I'm having issue with bed leveling too on this firmware. Creality CR-10 MAX

quarky42, 2021-05-01:
I'm trying to get 2.0.7_DW7.3.1 loaded but every time I try to flash the touchscreen, I get 0 files copied.   I've tried different SD and microSD(with adapter) cards.   I've formatted / ensured FAT32 format.  I can't seem to get the touchscreen flashed.

I am using the files I extracted from TM3D_SingleExtruderDwin_V3.1.7z and I am copying the whole DWIN_SET folder over to the root of the memory card so that there is just a DWIN_SET folder and all its contents on the card.  Not sure what is wrong here.

quarky42, 2021-05-01:
I2C_EEPROM still doesn't work on 2.0.7_DW7.3.1 for me and it had worked before on DW6.  I have a new breakout board that I'll swap into the printer in the future that has a new I2C_EEPROM on it, so I can check connections and try a new chip at that point to rule out hardware / wiring issues on my end.

For now I am using:

```
[env:10SPro_LPC1769]
platform = ${common_LPC.platform}
extends  = common_LPC
board    = nxp_lpc1769
build_flags = ${common_LPC.build_flags} -DMachineCR10SPro -DHotendAllMetal -DBedAC 
                                    -DABL_BLTOUCH -DHotendMosquito -DBondtech -DDirectDrive 
                                    -DSKR14Turbo -DFLASH_EEPROM_EMULATION -DSKR_2209 -DSKR_UART
                                    -DPROBING_MARGIN_LEFT=10 -DPROBING_MARGIN_RIGHT=0
```

That works for getting values to save and come back.  I haven't seen a .dat file on the microsd card for a long time.  Not sure where it's saving.

The default bltouch bed mesh probing is still messed up.  I have to set a left margin of 10 just to get the bltouch to the bed in the X direction.  I also need to set a clipclearance of 25mm just to get it onto the bed in Y.  I don't think this is right.

I wish I knew how to override values that were already set in code, for example, I would like to use:
-DPROBING_MARGIN_FRONT=15, but I get compile errors that it is already defined.  I've tried using ubuild_flags or -UPROBING_MARGIN_FRONT before the -D version of that flag, but that doesn't seem to work.  It would be nice to have more of my customizations in the platformio.ini file because they would be more self contained that way.  I would put my stepper motor current changes in there if I could.

InsanityAutomation, 2023-02-17:
Closing as this was all from an upstream bug

closed 2023-02-17 by InsanityAutomation