InsanityAutomation/Marlin #162: DW7 - Serial Port (Pronterface) connection not working

Frontier314, 2020-11-18:
### Description

DW7 version of CRX_NoFil_DW7 cannot connect to Pronterface.
Upon opening the connection, the serial port displays "External reset" and it is not possible to issue g-Code commands.

### Steps to Reproduce

1. Install CRX_NoFil_DW7
2. Connect to the CRX via Pronterface on COM port speed 250000
3. Issuing any g-Code command, Pronterface responds "[ERROR] Printer is not online." Pronterface "CONNECT" button does not change to "DISCONNECT" meaning that the USB serial connection is closed.

**Expected behavior:** [What you expect to happen]

To be able to connect to printer console via Pronterface and issue g-Code commands, as I could do with DW6.2.

**Actual behavior:** [What actually happens]

Immediately upon clicking "CONNECT", the connection is closed via external reset.

LMS0815, 2020-11-18:
same issue see: https://github.com/InsanityAutomation/Marlin/issues/154

InsanityAutomation, 2021-04-28:
Closing as duplicate

closed 2021-04-28 by InsanityAutomation