InsanityAutomation/Marlin #165: CrealityDwin_2.0 UI goes chinese after successful print. Machine needs reboot to print again.

Ericvf, 2020-12-02:
After a print completes successfully the UI becomes Chinese. Changing it back to English doesn't help, the UI doesn't respond to a lot of commands. I need to cycle the printer in order to get it working again. 

### Steps to Reproduce

1. Clone the repo, my branch is the CrealityDwin_2.0
2. Uncomment compiler directive "PLUS" since I use a CR-10s Pro.
3. Compile and flash
4. Print any file from SD card

**Expected behavior:** 

The print succeeds. It executes the ending G-code. Shows English prompt that the print was successful. Press OK and try printing another file. UI doesn't respond to the "Print" button. 

**Actual behavior:** 

The print succeeds. It executes the ending G-code. Shows Chinese prompt that the print was successful. Press OK and print another file.

Crash416, 2021-04-12:
I am having this same issue... has there been any update or fix?

Ericvf, 2021-04-13:
Not that I know of. I'm still having this issue. It doesn't happen for every print but it does keep happening. Very annoying.

InsanityAutomation, 2021-04-28:
Can you confirm if you see this in the current builds? I have not reproduced it here.

clue3030, 2021-04-28:
I had this issue on my Ender 6 but it clears up on it’s on in less than a second.

closed 2021-06-20 by InsanityAutomation