InsanityAutomation/Marlin #167: Error compiling for board Arduino Mega or Mega 2560 (CR-10 Max)

christianitus03, 2020-12-08:
### Description

When attempting to compile Marlin for my CR-10 Max, I get this message:
`Error compiling for board Arduino Mega or Mega 2560`.

### Steps to Reproduce

1. Open Marlin.ino in the Arduino IDE.
2. Uncomment line 19 of Configuration.h (`#define MachineCR10Max`)
3. Click the "Verify" button

**Expected behavior:** [What you expect to happen]
Marlin compiles normally.

**Actual behavior:** [What actually happens]
I receive the following error message after about one minute of compilation:

Arduino: 1.8.13 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/MarlinSerial.cpp:39:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/eeprom.cpp:24:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/inc/MarlinConfig.h:41:0,
                 from sketch/src/MarlinCore.h:24,
                 from sketch/src/MarlinCore.cpp:31:
sketch/src/inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/HAL.cpp:24:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/inc/MarlinConfig.h:41:0,
                 from sketch/src/MarlinCore.h:24,
                 from sketch/src/MarlinCore.cpp:31:
sketch/src/inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/eeprom.cpp:24:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/Servo.cpp:56:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/HAL_SPI.cpp:34:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:441:4: error: #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
   #error "PSU_ACTIVE_HIGH is now PSU_ACTIVE_STATE. Please update your configuration."
    ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/MarlinSerial.cpp:39:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/Servo.cpp:56:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/HAL_SPI.cpp:34:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
In file included from sketch/src/HAL/AVR/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/AVR/HAL.cpp:24:
sketch/src/HAL/AVR/../../inc/SanityCheck.h:2925:6: error: #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
     #error "PSU_CONTROL requires PSU_ACTIVE_STATE to be defined as 'HIGH' or 'LOW'."
      ^~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

#### Additional Information
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[My Marlin folder](https://github.com/InsanityAutomation/Marlin/files/5662284/Marlin.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[Screenshot, after selecting "Verify"](https://user-images.githubusercontent.com/39102269/101548663-4026a580-397a-11eb-8e12-cd5f6f9f77db.png)

DrDMoney, 2020-12-15:
PSU_ACTIVE_HIGH needs changed PSU_ACTIVE_STATE.

Also,
PSU_ACTIVE_HIGH FALSE needs changed to PSU_ACTIVE_STATE LOW.

InsanityAutomation, 2021-04-29:
Closing this one as its long since resolved, and not something enabled in any standard builds

closed 2021-04-29 by InsanityAutomation