InsanityAutomation/Marlin #172: CR-10S Pro v2: Error after finished print

Putaro2, 2020-12-29:
Printer: CR-10S Pro with BLTouch
Firmware: TM3D 10SProV2, 2.0.6_DW7

After my print is finished, i press the finished print button and one of two things happen:

1: Language changes to chinese. Printer doesn't print a new file.
2:Printer doesn't print a new file.

So i have to restart the printer, in order to print again.

I tried to flash the stable version DW6.2 but that didn't work. I cannot flash the current version again using avrdude.
Cura, Arduino and so on do not work either!

I have previously, successfully used the command:
`sudo avrdude -v -p atmega2560 -c wiring -P /dev/ttyUSB0 -b 250000 -D -U flash:w:/home/putaro/Desktop/10SProV2_BIL_DW7.hex:i`

When i try to use it again, i receive the following error:
```
avrdude: serial_baud_lookup(): Using non-standard baud rate: 250000avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.
```

Ericvf, 2021-04-13:
I'm having the same issue with DW7 on my CR-10s Pro v1. 
I've compiled the firmware itself.

InsanityAutomation, 2021-06-20:
As I have not been able to reproduce on current builds, I am going to close this. Please feel free to reopen if you are able to reproduce.

closed 2021-06-20 by InsanityAutomation