InsanityAutomation/Marlin #173: CR10S PRO DW7 problem with filament control

mclareniak, 2021-01-02:
Good day.
Excuse my weak English, I use a translator.
I own a Creality CR10S PRO 3D printer. I upgraded my printer and installed a BL-Touch. I used the FW original from creality for my printer with BL-Touch, specifically this CR-10sPro_1.60.8 BL.
I have found that the correct filament check is not working. That&#039;s why they told me in creality that I have an update fw to a newer version. Since I have a BL-Touch mounted, the offered fw was for the CR10S PRO V2, specifically this CR-10S ProV2_V1.70.0 BL. This solved my problem.
I wanted to test your fw for my printer. So I specifically uploaded this 10SPro_BLT_BIL_DW7.hex. BL-Touch didn&#039;t work for me with this fw. So I tried to load fw 10SProV2_BIL_DW7.hex. BL-Touch already works with this fw, but the filament control has stopped working.
Watch the video https://youtu.be/C4unX-KI1C4
Can you please advise me how I can solve my problem?
Thank you best regards Ziman.

Guilouz, 2021-01-03:
This is a known issue with DW7, use DW6.2 until it's fixed.

https://docs.google.com/document/d/1cuYWpDw8aPuLEXus3uNsgDIEAKEze3Z9-hIDMWcqMK8/edit

InsanityAutomation, 2021-04-28:
Im closing this as several releases since have resolved this issue.

Thanks

closed 2021-04-28 by InsanityAutomation