InsanityAutomation/Marlin #174: CR5 Pro Issue/request for marlin

danis101, 2021-01-02:
Hi!
Im struggling with cr5pro since part cooling fan in stock marlin cannot be controlled - 99% power fan off 100% fan on
I need to change sometimes those values so its pretty important thing. Creality as everyone know is hard to contact.
I wanted to fix it on my own but when I flash firmware LCD does not boot, I also dont know how to controll LEDS on this printer with are connected to HE2 port.
The board is creality v2.5.1 and doesn't fit to any other printers Ive searched
But back to topic, Ive measured voltages on the uotput of fan port and its seems to be correct, so actually I was thinking that is PWM problem

Anyways Ive compiled fresh marlin but cant get LCD to work so I found you guys
LCD in CR10 MAX is pretty similar but in cr5 there is different UI than on cr10MAX
anyway it does not even turning on with fresh installed marlin, the only one option atm is to wait for fixing it by creality or just make new software where those dwin lcd's and LED lights make me tons of troubles

Please help me

InsanityAutomation, 2021-04-27:
Looking over it, this machine seems to have the identical touchscreen. I don't currently have one to work with though. Typically a distributor such as Tiny Machines who plans to carry them will send me one to get working if there is enough interest however requests for this have been just individuals so far. It's likely this could be made to work, but I'd need a machine to work with.

danis101, 2021-05-12:
I could actually help as  much as I can. This machine seems to be great, some things has to be fixed such as firmware and hotend for metal one (why the hell they are limiting this to 240C since it  has enclosure) 
I Flashed working marlin but can't figure out how to control led bar through dwin - Dwin is a kind  of wall for me :/

InsanityAutomation, 2023-02-17:
The CR5Pro HT was added awhile ago. Should be mostly identical, and will probably just work. I dont have both versions to verify, but im marking this done unless i hear otherwise.

closed 2023-02-17 by InsanityAutomation