InsanityAutomation/Marlin #179: file name CR10S_BLT_LR_DW7 what does LR and what file i should use on stock cr10s with BLtouch

cisco150, 2021-02-17:
Hi Guys in this file name CR10S_BLT_LR_DW7.hex from Marlin-CrealityDwin_2.0  what does LR mean and what file should i use with stock cr10s with bltouch need help looking into it and what different from this file CR10S_BLT_DW7.hex

quarky42, 2021-03-14:
If you open up the platformio.ini file you can look at the various builds in there.  Many of them now have LR as part of one of their tags.

That tags in the hex file usually have to do with options that are enabled, so look at the build_flags line and see what you think.  (I'm not being flippant, I myself am not 100% certain, so look for yourself and see if you agree or disagree with this next bit.)

I think LR is Loss Recovery as in Power Loss Recovery.   The only Creality printers that don't have LR in their names (at least that I saw) also did not have the Power Loss Recovery flag set.

InsanityAutomation, 2021-04-27:
PLR is standard in all the configs it fits on. Ive added the LR tag (Lerdge runout sensor) to the readme files for reference, its a replacement thats opposite stock polarity.

quarky42, 2021-04-27:
I gladly stand corrected and thank you for explaining.

I forgot about the Lerdge sensor... Had seen it before but completely forgot.

closed 2021-04-28 by InsanityAutomation