InsanityAutomation/Marlin #182: CRX_BLT_BIL_NoFil_DW7 LCD Still shows Manual Leveling options

colwellkr, 2021-02-20:
Hello!

I have upgraded my CR-X with a BLTouch sensor and flashed it with the CRX_BLT_BIL_NoFil_DW7.hex build of Marlin 2.0.

Homing works great with the BLTouch sensor, but when I try to go into the leveling menu, it still shows only the Manual Leveling menu from the stock CR-X. When I go into the menu, the printer homes itself and then tapping on any of the locations just initiates another home.

I also tried flashing CRX_BLT_UBL_NoFil_DW7.hex and get the same result. Same thing happens when manually initiating a G28. It seems that something isn't quite set right in the configuration?

I'm trying to build the firmware myself as I've heavily customized my CR-10S and basically upgraded it into a 10S Pro, but I'm not super familiar with the customization you have done and how to work with touch screens. I see a 7zip file in there for CR-X ABL touch screens, but not sure how to use it.

I really appreciate the help!

colwellkr, 2021-02-20:
Update for anyone coming after me, the solution ended up being that I had to flash the LCD files. Instructions are here:

https://youtu.be/SBX30GmM3Qo

closed 2021-02-20 by colwellkr

LMS0815, 2021-02-20:
Can you **please** share the CR-X config files as there are missing everywhere.

InsanityAutomation, 2021-02-21:
@LMS0815 they're right in the main config file with the flag comparisons with everything else? Been there all along.

colwellkr, 2021-02-22:
^^ I found that I didn't have to build my own version. But if you do, there are just a bunch of flags you enable to set it up for the CR-X and customize to your printer.