InsanityAutomation/Marlin #183: Display message error when doing Auto PID on CR-10 Pro

DaKrampus, 2021-02-23:


InsanityAutomation, 2021-04-27:
This looks similar to an issue with pronterface sending M0 commands to machines at certain points. That would call the dialog. 
The steppers will disable themselves after a couple minutes however if you want to move it by hand sooner, under Settings you can disable the steppers immediately. I have not seen this reproduced without pronterface connected specifically, no other terminal causes this to my knowledge.

DaKrampus, 2021-04-27:
Sorry, I forgot to mention that i only hooked up pronterface after i had had the issue. So I do not think it is linked to pronterface. 
Maybe it did change some things in the background, but the main issue, having the filament change requester showing up on Auto PID is exactly the same.
To fix it for my usage, I now downgraded from the 7.2 version to 6.2 which does work fine. (Old things are not allways bad ;-)
Keep up the good work

Luc

InsanityAutomation, 2021-04-27:
I ran through the PID tune a few more times this morning and didnt see this come up, can you check if 7.3 still exhibits this issue?

DaKrampus, 2021-04-27:
I will, but i am on a job assignment abroad and will only be back home on sunday evening.. I will check as quickly as possible on monday.. (sorry for the delay)

Luc

InsanityAutomation, 2021-04-27:
No rush, been away from this for awhile because of the chaos the pandemic brought to my every day life but trying to make sure everything that was left open is cleaned up at this point. Probably alot here I solved a long time ago but never touched base with anybody.

InsanityAutomation, 2021-05-24:
Sounds good! Between my testing and positive reports, ill close this. If there are still any issues just let me know and it can always be reopened.

closed 2021-05-24 by InsanityAutomation