InsanityAutomation/Marlin #186: cr-10 v3: exrusion is reversed

amrnablus, 2021-03-27:
### Description
It seems that the direction of the extruder motor is flipped on cr10-v3 (i'm aware i can flip the motors wire but it's a pain in the butt)
### Steps to Reproduce

install CR10V2_BLT_DW(X)
heat the extruder
move access -> extruder 

**Expected behavior:** [What you expect to happen]
extruder goes backwards
**Actual behavior:** [What actually happens]
extruder should go forward

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2021-04-27:
If my memory is correct, the V3 has the Titan extruder preinstalled which reverses the direction. I can look at adding that configuration to the pre built list for the next release, but it is correct that it is reverse of the bowden version so inverting that file would cause the V2 bowden to be backwards.

InsanityAutomation, 2021-06-20:
CR10V3 default file now included in the mix for the next release.

closed 2021-06-20 by InsanityAutomation