InsanityAutomation/Marlin #188: Unable to connect after upgrading to E5P_UBL_SlntH_DW7.2

Iveusse, 2021-04-14:
### Description

After updating to E5P_UBL_SlntH_DW7.2 (whatever the previous version), there's no way to connect to the printer from Octoprint or Pronterface. Moreover, printer is stuck on the firmware loading screen.

### Steps to Reproduce

Build E5P_UBL_SlntH_DW7.2 using PIO build task (without any modification in the Configuration.h)
Flash firmware from Octoprint Firmware Updater plugin
Try to connect to the printer or use the printer from screen.

**Expected behavior:** Octoprint connects to the printer. Printer interface works.

**Actual behavior:** no way to connect from Octoprint. Printer stuck on firmware loading screen.
#### Additional Information

I'm still able to flash the printer from Octoprint so I was able to move to the E5P_BIL_SlntH_DW7.2 which works well.
Configuration.h and Configuration_adv.h are the original one from this branch.

InsanityAutomation, 2021-04-27:
Can you confirm if this is still an issue with 7.3? Some 7.2 builds were exceeding RAM limits.

Iveusse, 2021-04-28:
That works ! Thank you

closed 2021-04-28 by Iveusse