InsanityAutomation/Marlin #189: SV02 Support

Wob76, 2021-04-26:
Hi,

I'm not sure if you are interesting in incorporating this or not but just an FYI, I have started making some basic changes to make your code work on the Sovol SV02 (https://sovol3d.com/products/sovol-sv02-dual-extrusion-3d-printer), it seems Sovol have based it on the CR-X, smaller build volume but most (if not all) the electronics seem to be the same. My screen is lacking a speaker, not sure if the CR-X has one? I have added one as I found the lack of feedback from the screen annoying.

I have it running successfully, but still going through it make sure I don't have any bugs. I have tried to copy your formatting, and where I have wanted a setting enabled I have used if statements to ensure it doesn't affect other builds.

I have kept changes to a minimum at this stage, there may still be some features that it doesn't play nicely with, but I am still working through those.

At this stage I have customised the ABL version of the CR-X Screen to "rebrand" it from Creality to Sovol, and it does require a remount of the screen as the stock Sovol setup is in portrait mode, I'll link in a STL to my mount once it is uploaded.

Anyway my changes can be found at https://github.com/Wob76/Marlin/tree/CrealityDwin_2.0

InsanityAutomation, 2021-04-27:
At a glance it seems nearly identical in most ways! I may need to take a look at this machine a bit more as things settle down. Im hearing more requests for the CR5 and the Ender 6 at the moment though (I know, too many confusing names from creality...)

Wob76, 2021-04-27:
I tried to keep changes to a minimum, it is mostly just adjustments for build volume.

I did define "CrealitySilentBoard" as this machine comes standard with the Creality v2.2.1 board, is is defined in the machines list then I just defined "MachineCRX".

I did limited the baudrate to 115200 to play it save, but a quick test and it does seem to work at 250000.

Apart from that Steps were different (stock they are 93) and the BED Temp Sensor was different.

The stock firmware did have different values for things like Feedrate, Acceleration and Jerk, you values were a little more aggressive, but my though process was if the larger CR-X can handle them they should be fine and this could keep the changes required to a minimum. so far print quality has been fine with the new values.

The SV02 is work a look, it is great value if you don't need the build volume of the CR-X and so far it looks like a very similar machine.

I have made note of a few bugs, some of which have been reported upstream in Marlin, are you OK if I dump them all into a single "issue" or send them in a email rather than opening a bunch of issues?

InsanityAutomation, 2021-04-27:
Single issue is fine. If there is an upstream report with more information you're aware of, try to include it if you can.

Wob76, 2021-04-27:
Done and Done

BigRigCookin, 2022-09-18:
So was this ever finished? This firmware that I have from sovol is in need of desperate repair. The Sovol SV02 Printer

Wob76, 2022-09-23:
I've not been doing much 3D Printing stuff in the last 12 months, I did open ticket #190 and was going to report back. I have my slightly modified version of this firmware running on my SV02, I think most of the bugs were ironed out in the update mentioned in the other ticket, no show stoppers anyway. My repository is below, last updated June 2021. The LCD firmware has updates for the Sovol Logo and mostly just changes to match the size difference to the CR-X, you should be able to see it in the last 2 commits.

https://github.com/Wob76/Marlin/tree/SV02_Dwin_2.0