InsanityAutomation/Marlin #190: Firmware Bugs

Wob76, 2021-04-27:
As mentioned I have noted a few bugs while testing the implementation on the SV02.

Note: I am printing from Octoprint

I'll log a separate issue for a M600 bug as I have a fair bit of info for that alone.

**Tool Change** - The Tool change does not retract the filament during a tool change, only loads, upstream bug has been reported here (https://github.com/MarlinFirmware/Marlin/issues/20652 and https://github.com/MarlinFirmware/Marlin/pull/21142). I have disabled the toolchange for the moment with M217 R0 W0 , I am still using a prime\wipe tower but would like to use the toolchange one I get to looking at a purge bucket.

Finished Print - When finished it shows end of print screen on the LCD (finished in octoprint), when you hit "Finish" it homes and shows "Print Aborted". //action:cancel and //action:notification Print Aborted is seen in the terminal. Not a huge issue if you just ignore the screen, but would be nice to Finish without homing and without the error.

**LCD Bugs**
Tools - Refuel - When changing the amount to be extruded for T1 the keypad is clipped, too far to the right and too far down
Tools - Refuel - Temperature Warning, the minimum Temp, shown in red, is misaligned, needs to be shifted right.
Temp - Cooling - The LCD changes to show the Fan as being turned on, fan does not get triggered, LCD shows fan as on until you press it off.

I have fixed one of these 3 (the second) LCD bugs when "rebranding", I had to do this by binary editing the .CFG file as when I try and edit the project with the current DWIN DGUS v8 tool it seems to mess up the output, do you have a recommended version for the project?

**Excessive Beeps on Filament Change** -  The LCD beeps 5 or 6 times, despite being set to 2 (or even 1) in the Marlin Config, you can see multiple "***CPlay Tone***" in the terminal so it seems to be an issue in Marlin somewhere.

**SD Card Removed** - I see lots of Card Removed*** being seen in the terminal regularly, note the missing "***" at the start. I did see mention of some changes to account for broken card detection switches, I don't actually have an SD Card inserted since I am using Octoprint, this may be normal behaviour, but since it is clipped I thought it worth mentioning

**Excessive Waits in the Terminal** - with this build you can see excessive "wait" responses in the terminal, even more than the temp reports, I can and have filtered them, but just wondering if this can be stopped or reduced or maybe a requirement of the LCD?

Thanks,
Beau

InsanityAutomation, 2021-06-20:
Finish print behavior has been changed

I havnt looked into the upstream tool change bugs yet. It seems the PR is still sitting. Ill add that to my list and if it hasnt been handled after im done with the next few tasks in line ill dig into it.

DGUS version used is 7.383 which is matching the firmware Creality originally loaded. 8.x and some of the added controls can only be used with a firmware update, which has a risk of bricking displays.

Refuel issues fixed

Cooling ill look into next go round

Excessive beeps is mainly due to the code as its written based on a simple buzzer, not the wav file play the LCD uses. Typically the display will ignore additional play sound requests while one is active so its not a problem aside from in the serial display, but its definitely not optimal. Couple be improved in the future but thats something for another day.

SD Card removal sounds like the old string issues dealing with an overflowing buffer. Should be gone now.

Waits are part of the host keep alive feature upstream, nothing in the touchscreen code changes those.

Wob76, 2021-06-21:
Thanks for the updates, I'll try updating soon and test the changes out.