InsanityAutomation/Marlin #191: M600 Not offering Purge Options

Wob76, 2021-04-27:
When you initiate a M600 the print will resume after the load without offering a Purge More\Continue Option, It also seems that the LCD is treating this as a Filament Runout as that is how all the LCD prompts are worded.

In the Terminal you can see the Action Prompts being sent from Marlin, but it seems to action itself. I have disabled the "Action Prompts" in Octoprint to ensure that it wasn't sending any actions.

After the M600 you receive a prompt for the pause with only a dismiss option (Retract take place regardless of response)
You then receive a Second prompt for the Nozzle Park, Shows as a runout, Yes to Change filament or No to cancel.
Yes and the filament is loaded and then the print head returns to it position.

Here are the logs of the process as captured in Octoprint, noting you never see the "PurgeMore\Continue" Prompts and no user action causes the action:resumed, unless it is caching a previous prompt. I will try and capture a video of the process using when using the LCD.

```
wait
M600
//action:paused
//action:prompt_end
//action:prompt_begin Pause
//action:prompt_button Dismiss
//action:prompt_show
 T:180.02 /180.00 B:33.36 /0.00 @:43 B@:0
echo:busy: processing
 T:179.98 /180.00 B:33.33 /0.00 @:43 B@:0
echo:busy: processing
 T:179.79 /180.00 B:33.23 /0.00 @:46 B@:0
echo:busy: processing
 T:179.79 /180.00 B:33.26 /0.00 @:45 B@:0
X:50.00 Y:10.00 Z:65.90 E:-4.00 Count X:4000 Y:800 Z:26369
echo:busy: processing
 T:179.88 /180.00 B:33.10 /0.00 @:44 B@:0
echo:busy: processing
 T:179.98 /180.00 B:33.05 /0.00 @:43 B@:0
echo:busy: processing
 T:180.10 /180.00 B:32.99 /0.00 @:41 B@:0
echo:busy: processing
 T:180.29 /180.00 B:32.89 /0.00 @:39 B@:0
echo:busy: processing
 T:180.34 /180.00 B:32.97 /0.00 @:39 B@:0
echo:busy: processing
 T:180.36 /180.00 B:32.89 /0.00 @:38 B@:0
echo:busy: processing
 T:180.31 /180.00 B:32.71 /0.00 @:39 B@:0
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
==onUserConfirmRequired==
***CPlay Tone***
***CPlay Tone***
 T:180.55 /180.00 B:33.49 /0.00 @:34 B@:0
echo:busy: paused for user
***CPlay Tone***
***CPlay Tone***
***CPlay Tone***
***CPlay Tone***
 T:179.38 /180.00 B:33.41 /0.00 @:54 B@:0
echo:busy: paused for user
***CPlay Tone***
 T:180.02 /180.00 B:32.50 /0.00 @:42 B@:0
echo:busy: paused for user
 T:179.92 /180.00 B:32.73 /0.00 @:43 B@:0
echo:busy: paused for user
 T:179.85 /180.00 B:32.66 /0.00 @:44 B@:0
echo:busy: paused for user
 T:179.92 /180.00 B:32.68 /0.00 @:43 B@:0
echo:busy: paused for user
 T:179.96 /180.00 B:32.71 /0.00 @:42 B@:0
echo:busy: paused for user
 T:180.05 /180.00 B:32.68 /0.00 @:41 B@:0
echo:busy: paused for user
  *******RTS_HandleData********
recdat.data[0] ==1
recdat.addr ==4190
== Checkkey==
22
Card Removed***

 No Filament
Resume Yes during print
M876 Responding PROMPT_FILAMENT_RUNOUT_CONTINUE
 T:179.88 /180.00 B:32.53 /0.00 @:44 B@:0
echo:busy: processing
 T:180.10 /180.00 B:32.66 /0.00 @:41 B@:0
echo:busy: processing
 T:180.10 /180.00 B:32.60 /0.00 @:41 B@:0
echo:busy: processing
 T:180.10 /180.00 B:32.66 /0.00 @:41 B@:0
echo:busy: processing
 T:180.05 /180.00 B:32.53 /0.00 @:42 B@:0
echo:busy: processing
 T:180.02 /180.00 B:32.42 /0.00 @:42 B@:0
echo:busy: processing
 T:180.00 /180.00 B:32.40 /0.00 @:42 B@:0
echo:busy: processing
 T:179.96 /180.00 B:32.47 /0.00 @:43 B@:0
echo:busy: processing
 T:179.96 /180.00 B:32.29 /0.00 @:43 B@:0
echo:busy: processing
 T:179.96 /180.00 B:32.27 /0.00 @:43 B@:0
echo:busy: processing
 T:179.98 /180.00 B:32.16 /0.00 @:42 B@:0
echo:busy: processing
 T:179.94 /180.00 B:32.16 /0.00 @:43 B@:0
echo:busy: processing
 T:179.98 /180.00 B:32.11 /0.00 @:42 B@:0
echo:busy: processing
 T:179.85 /180.00 B:32.01 /0.00 @:45 B@:0
echo:busy: processing
 T:179.75 /180.00 B:32.03 /0.00 @:46 B@:0
echo:busy: processing
 T:179.38 /180.00 B:32.01 /0.00 @:52 B@:0
echo:busy: processing
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
//action:prompt_end
 T:179.40 /180.00 B:32.03 /0.00 @:51 B@:0 W:?
 T:179.27 /180.00 B:31.95 /0.00 @:53 B@:0 W:1
 T:179.23 /180.00 B:31.98 /0.00 @:53 B@:0
echo:busy: processing
 T:179.21 /180.00 B:31.90 /0.00 @:54 B@:0 W:0
//action:notification Print Paused
 T:179.13 /180.00 B:31.90 /0.00 @:55 B@:0
echo:busy: processing
 T:179.31 /180.00 B:31.93 /0.00 @:52 B@:0
echo:busy: processing
 T:179.63 /180.00 B:32.01 /0.00 @:48 B@:0
echo:busy: processing
 T:179.90 /180.00 B:31.90 /0.00 @:45 B@:0
echo:busy: processing
//action:resumed
//action:notification TM3D SV02 Ready.
ok P7 B3
wait
```

InsanityAutomation, 2021-06-20:
While messaging isnt as good as it could be, next build here has the ability to purge now. This will get even better as time goes on.

closed 2021-06-20 by InsanityAutomation

Wob76, 2021-06-21:
Thanks for all you work, I'll look to give this a spin soon.