InsanityAutomation/Marlin #194: Babysteps issue

RS0ares, 2021-04-29:
Printer: CR10S Pro V2
Firmware: 10SProV2_50W_ET_DW7.3

 Today I installed a mosquito hotend with DDX PH4 300°C 24v 50W for CR-10(S)Pro/Max and i updated the firmware of my printer.

But I tried use the babysteps to level my bed, but the babysteps doesn't work correctly.

Ericvf, 2021-04-29:
I'm having an issue with Babystepping as well. See this issue:
https://github.com/InsanityAutomation/Marlin/issues/193

InsanityAutomation, 2021-04-29:
Closing as a duplicate of #193 however the bleeding branch should be resolved at the moment. Ill be generating more hex files after some more testing, and figuring out what @Ericvf is running into.

closed 2021-04-29 by InsanityAutomation

RS0ares, 2021-04-29:
OK,I will wait for the hex files to update my printer.
👍

InsanityAutomation, 2021-04-29:
[10SProV2_50W_ET_DW7.3.1.hex.zip](https://github.com/InsanityAutomation/Marlin/files/6398835/10SProV2_50W_ET_DW7.3.1.hex.zip)

Give this a try

Ericvf, 2021-04-29:
What a champ! Can confirm babystepping is resolved for me now. 👍

RS0ares, 2021-04-29:
Later, I'll try the file, then I'll tell you how it was.
👍

RS0ares, 2021-04-29:
Thanks a lot mate!
It's working 😃