InsanityAutomation/Marlin #196: Undefined References to setup and loop

griffout, 2021-05-11:
Hello, 

No matter what I do, I keep getting this error message. I've put about 6-8 hours into troubleshooting this and I don't know if it is a bug or what. What should I do? The error message reads.... 

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\aaron\AppData\Local\Temp\ccdwVOAC.ltrans0.ltrans.o: In function `main':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

InsanityAutomation, 2021-05-24:
The windows store version of the Arduino IDE is always even more broken than the standalone installer. In either case, Arduino IDE is no longer supported and VS Code with PlatformIO is the recommended method for installation.

griffout, 2021-05-25:
> The windows store version of the Arduino IDE is always even more broken than the standalone installer. In either case, Arduino IDE is no longer supported and VS Code with PlatformIO is the recommended method for installation.

Thanks! I got it to install but now it won't connect to Simplify3d. I didn't see a hex file for this. Is there a hex file specifically for the Creality3d CR-10S PRO V2 with the E3D Hemera hot end? I normally just troubleshoot things until I figure it out but, I've been at this for about 2 weeks and have made zero headway...