InsanityAutomation/Marlin #197: Ender 6 settings enhancement and correction

Deed01, 2021-05-13:
### Description

1. ASCII Serial input is Default=4 in configuration_ADV could be 16 like MachineEnder3V2
2. SDIO_SUPPORT is wrongly set
3. Machine NAME not defined in version.h
4. Force 128 buffer size in serial Rx for USART (otherwize 64 is default and not enought with arcwelder)

### Steps to Reproduce

**Configuration_ADV.h** 

**1. Change this**

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#if ANY(MachineCR10Orig, SKRMiniE3V2, MachineEnder3V2) //melzi has more ram than a 2560
  #define BUFSIZE 16
#elif ANY(SKR13, SKR14, SKR14Turbo, SKRPRO11)
  #define BUFSIZE 8
#else
  #define BUFSIZE 4
#endif

**To this (adding MachineEnder6 next to MachineEnder3V2**

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#if ANY(MachineCR10Orig, SKRMiniE3V2, MachineEnder3V2, MachineEnder6) //melzi has more ram than a 2560
  #define BUFSIZE 16
#elif ANY(SKR13, SKR14, SKR14Turbo, SKRPRO11)
  #define BUFSIZE 8
#else
  #define BUFSIZE 4
#endif

**Configuration.h** 

**2. Change this**
#if ENABLED(MachineEnder3V2, MachineEnder6)
  #define SDIO_SUPPORT
#endif

**To this (ENABLED to ANY)**

#if ANY(MachineEnder3V2, MachineEnder6)
  #define SDIO_SUPPORT
#endif

**Version.h** 

**3. Change this**
#elif(ENABLED(MachineS5))
#define CUSTOM_MACHINE_NAME "500 SuPeR"
#elif ENABLED(MachineCR2020)
  #define CUSTOM_MACHINE_NAME "TM3D 2020"
#endif

**To this (add MachineEnder6 section) **

#elif(ENABLED(MachineS5))
#define CUSTOM_MACHINE_NAME "500 SuPeR"
#elif ENABLED(MachineCR2020)
  #define CUSTOM_MACHINE_NAME "TM3D 2020"
#elif(ENABLED(MachineEnder6))
#define CUSTOM_MACHINE_NAME "TM3D Ender6"
#endif

**Platformio.ini**
**4. Change this**
  ${common.build_flags} -std=gnu++14 -DSTM32_XL_DENSITY -DTEMP_TIMER_CHAN=4 -DMachineEnder6 -DABL_BLTOUCH -DPOWER_LOSS_RECOVERY -DABL_UBL

**To this (defined USART_RX_BUF_SIZE) **
  ${common.build_flags} -std=gnu++14 -DSTM32_XL_DENSITY -DTEMP_TIMER_CHAN=4 -DUSART_RX_BUF_SIZE=128 -DMachineEnder6 -DABL_BLTOUCH -DPOWER_LOSS_RECOVERY -DABL_UBL

If not defined USART_RX_BUF_SIZE is set to 64 meaning the max Gcode command received by serial is 64 characters (even if you have #define MAX_CMD_SIZE 96 meaning marlin can process up to 96 Gcode comman lenght => consequences the gcode file will be processed using SDCard but failed using serial)
Found this using arcwelder on Cura and octoprint

To test it 
1. send from octoprint terminal to marlin
G3 X137.523 Y127.650 I14.9500001 J111753.77300001 E67.67030001 F600 using octoprint

2. answer form marlin
ecv: echo:Unknown command: "37.523 Y127.650 I14.9500001 J111753.77300001 E67.67030001 F600"
Recv: ok P15 B15

only the last 64 characters are received

By setting USART_RX_BUF_SIZE=128, the command is processed

InsanityAutomation, 2021-05-24:
Long commands for sure! I dont have an Ender 6 yet so while this is of interest I have to way to test. I intend to sit down and get the Ender 6 and CR5 Pro added at some point but will need feedback from people with hardware. I will get to it at some point in the near future though.

InsanityAutomation, 2021-06-12:
FYI, thanks to Ben Liu and Tina / Xu from Creality, I got an Ender 6 here and am going through it now! Next update will include it. No CR5 or Sermoon yet, but hopefully soon to add those as well.

InsanityAutomation, 2021-06-20:
Closing this as I pack up the Ender 6 builds...

closed 2021-06-20 by InsanityAutomation

notfixingit3, 2021-09-12:
Did these changes not make it into the current code base?

InsanityAutomation, 2021-09-13:
It's in the development code that is kept offline nowadays due to a specific bad actor who was taking credit for fixes posted there in his own releases that were not attributing where he was taking the code from while posting all over his rather sizable Facebook group... Everything gets updated when I'm actually putting a release out to prevent that kind of theft.

notfixingit3, 2021-09-13:
oh that sucks, stupid people... I need to make a bondtech build for ender6 with a different display (creality was out of replacement touch screens, so they sent me one from a CR10 until they get more stock) so I was trying to figure out if these where already in or not or if they shouldn't be when I attempt to compile. This makes me worry there's other stuff I might be missing that was part of the current build though?