InsanityAutomation/Marlin #202: U8glib issues when compilling to CR10 MAX Hemera

typhus94, 2021-06-05:
### Description

Hello, I have a CR10 MAX that I upgraded with an Hemera extruder from E3D, and following the tutorial from Teaching Tech I installed this version of Marlin. I tried to update it to make the filament runout sensor work, but now I cant compile the firmware to the printer, and I keep getting error messages about the u8glib library being the issue. My coding knowledge is basically 0, so I feel completely lost. I have tried both Arduino IDE 1.8.15 (the latest version I can download) and VS Code with Platformio. I have uninstalled and installed back again both a few times. 

In Arduino I get a few of this kind of errors: 
"C:\Users\Ander\AppData\Local\Temp\arduino_modified_sketch_74667\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
C:\Users\Ander\AppData\Local\Temp\arduino_modified_sketch_74667\src\lcd\dogm\HAL_LCD_class_defines.h:131:13: error: 'init' is not a member of 'U8GLIB'
     U8GLIB::init(&u8g_dev_ssd1309_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);
             ^~~~"
And in Platformio, in the Configuration.h it starts with an "#include <U8glib.h>" but it cannot open the the source file U8glig.h

I have also tried to read all kind of tutorials but I feel very lost with this kind of coding. 

Thanks in advance

### Steps to Reproduce

-Download the "CR10 MAX marlin Insanity Automation.zip" 
https://drive.google.com/file/d/1Z4Pqnwn7pHoKIrskOtAKteqI-T5D_nsD/view?usp=sharing
-Open Configuration.h in Arduino IDE, and compile.
-Open the project in Platformio and build it.

InsanityAutomation, 2021-06-20:
At a glance, the configuration file doesnt have any of the machine options setup. Arduino IDE definitely wont be able to do anything with it. Were you using one of the predefined PIO environments?

InsanityAutomation, 2021-06-20:
Please see the discord links in the branch ReadMe for custom compiling support. Many more people there can help much easier and faster than just waiting till I circle around to this queue.

3D Printing Discord - https://discord.gg/MZrsgVU
Marlin Discord - https://discord.gg/n5NJ59y

closed 2021-06-20 by InsanityAutomation