InsanityAutomation/Marlin #204: CR-10S Pro V2 E3D Hemera Hex File?

griffout, 2021-06-11:
Would it be possible for you guys to upload a hex file for the Creality CR-10S Pro V2 with the E3D Hemera Volcano extruder/hot end upgrade? That would be greatly appreciated, thank you!!

tmitch205, 2021-06-28:
I agree... I actually emailed TinyMachines last night to see if they happened to have one. I have bought 3 Hemera's from them... Installed one a couple months ago on a CR-10S Pro V2 and had no issues with the firmware that was current at that time... I did the same install on an identical machine yesterday and this version of the firmware has been a pain in the rear. If I cant get support here for the issues I wasnt able to resolve, then the 3rd one I am going to install on one of my machines with the Duet board... they are so much easier to deal with in terms of firmware changes and config files.

idesignstuff, 2021-09-29:
So I have a CR10S Pro with a Hemera and Volcano. I also still have an inductive ABL sensor. I'm not sure what firmware needs to be different. The big changes are around the offset from the nozzle to the ABL sensor, and you can do that with M851. https://marlinfw.org/docs/gcode/M851.html I've run this setup for months with very few failed builds. Love it.

However...

This latest UBL version of TM Marlin for the V1 CR10S Pro seems to have broken the ABL. It seems to stop and wait for the nozzle to get to temp for like a minute or two for each stop on the ABL grid, and going somewhat randomly from point to point. It also seems to not hit all the points and quits early.

I'd like my printer to manage the temps with PID like for everything else, but it seems to forget that while in ABL.