InsanityAutomation/Marlin #205: 10SProV2_BIL_DW7.4 Z-axis inverted

Zizor94, 2021-06-27:
As the title says, i flashed the firmware onto my CR10S Pro V2, then flashed the other files to the touchscreen card and thought everything was fine.
When i tried raising and lowering the Z-axis the controls were inverted (X and Y were normal though). Could you please take a look at this and fix it?

tmitch205, 2021-06-28:
I am having the same problem with 2.0.9_DW7.4 - I went in and fixed it and flashed it again.... but this revision has given me nothing but issues. Putting in a ticket for those other issues now.

chappy1978, 2021-06-28:
Same here built a custom version and z is inverted.

tmitch205, 2021-06-28:
I installed the Hemera on one machine a couple months ago and updated to the then current firmware and had no issues what so ever... I just did the exact same install on another identical machine with this firmware and not only was there the Z issue... the nozzle location on the bed is wrong. It's moved forward on Y a good 15mm and to the left on X a good 5mm... Problems like this on RepRap for my duets are quickly fixed on the fly... I am sure I can figure this out with time, but at 30 to 45 minutes for each change, sketch, upload, and test for Marlin, I decided to email TinyMachines and see if they happen to have a hex that they can just send me. Is there a way to copy firmware from one machine that is working perfectly to flash to another identical machine?

mercury049, 2021-06-28:
Same issue on a CR10S Pro V1.

InsanityAutomation, 2021-06-28:
Pinned the other duplicate issue, a bad , botched a few autobuilds so bad files were up for a day... Replaced now!

closed 2021-06-28 by InsanityAutomation

chappy1978, 2021-07-03:
Thanks for the fix. Great work as always.