InsanityAutomation/Marlin #206: 10SPro_BLT_BIL_DW7.4.hex Z Axis inverted for 10SPro V1

mercury049, 2021-06-28:
### Description

The Z axis is inverted on latest release for V1

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2021-06-28:
The corrupted files from a bad , were replaced. Please grab the current set and reflash! They sat for about a day before we noticed the autouilder botched up on us :/

Ill leave this open and on the top for a bit so anyone else who comes across it from the files as were staging the release tag can find the solution!

tmitch205, 2021-06-28:
That's awesome. Thank you for how quickly this was handled. For the Y issue, I wound up setting the endstop min to -15. That now puts my nozzle on the middle of the bed when jogging to 150mm.

InsanityAutomation, 2021-06-28:
> 
> 
> That's awesome. Thank you for how quickly this was handled. For the Y issue, I wound up setting the endstop min to -15. That now puts my nozzle on the middle of the bed when jogging to 150mm.

Sounds like the hemera mount youre using shifts the nozzle position a bit. M205 would likely have resolved it without reflashing if shifting the  origin did the trick. Probe offsets are already on the screen, but if many hotend mounts cause the origin to need adjusting, home offsets could be added as well at some point.

tmitch205, 2021-06-29:
> Sounds like the hemera mount youre using shifts the nozzle position a bit. M205 would likely have resolved it without reflashing if shifting the origin did the trick. Probe offsets are already on [...]

That could be very useful. For me, I went ahead and exported to a hex so that the next one I do is ready to go. :) another idea in lieu of that would be reference to the mount that the code is written for.

InsanityAutomation, 2021-06-29:
> 
> That could be very useful. For me, I went ahead and exported to a hex so that the next one I do is ready to go. :) another idea in lieu of that would be reference to the mount that the code is written for.

TBH Im not even sure!! lol I dont have a hemera here, so I just put in the values that Chris at tiny machines fed me for what he was using. Ill see if there is a link to it or if its an in-house design

closed 2023-02-18 by InsanityAutomation